sebanc/brunch #278: Boot times

Pumpino, 2020-06-07:
How long should Brunch take to boot?

On my main PC (HP ProDesk 400 with i3, 8GB of RAM and a new Crucial MX500 256GB SSD), CloudReady boots in 10 seconds, but Brunch takes over one minute. I'm wondering if that's typical and if there's anything I can do to change it.

sebanc, 2020-06-07:
Hi,

brunch boot time is long because of the initramfs which is the main piece of brunch (it allows the different configuration options and updates notably):
- a delay is needed for all block devices to be mounted by the kernel.
- then it verifies if changes have been made in options or on partition 7 (brunch update)
- ui display is also delayed to be more compatible (notably for accelerometer and android apps support). You can try "options=no_ui_delay" to have a slighly faster boot.

Cloudready does not use an initramfs which speed up things and they have no accelerometer / android apps issues as they do not support those features.

sebanc, 2020-06-12:
Explanation has been pinned in issue #304

closed 2020-06-12 by sebanc