sebanc/brunch #279: Webcam lo longer working after upating to M81 recoery image

saghul, 2020-06-07:
Hey there!

I've been using this with success on a 2013 rMBP and it's great, thanks for putting it together!

The only bug I currently face is that Android apps don't work with the camera. The camera does work in the browser.

Yesterday I noticed the M81 recoery image was up for rammus so I gave it a shot together with the latest brunch release and now the webcam doesn't work on the browser either.

Does this ring a bell? Any extra information I can provide to help you debug?

Thanks again for such a nice project!

sebanc, 2020-06-07:
Hi,

It appears that further to some changes in the google camera framework, some using "rammus" recovery image have seen their camera working in r81 and some have their camera not working anymore.

I am working on improving rammus camera support but until it is fixed could you try "reef" recovery image ?

saghul, 2020-06-07:
Will do, thank you!

saghul, 2020-06-08:
I gave reef (r81 recovery) but it now won’t boot. I see some boot text, then the screen goes blank and reboots. It happens to fast to see anything.

Any other thing I could try?

sebanc, 2020-06-08:
Your macbook has 4th gen intel, right ?

saghul, 2020-06-08:
Hum, might be a 3rd gen come to think of it!

saghul, 2020-06-08:
Looks like I need to use samus, oops.

sebanc, 2020-06-08:
Indeed try samus, I thought it was 4th gen but if it is 3rd gen reef will not work.

saghul, 2020-06-08:
Ok, so samus boots, but the camera still refuses to work. At least is somewhat different because Android apps don't crash when trying to use it.

sebanc, 2020-06-12:
It is weird, so far samus worked for everyone who had a usb camera (which seems to be your case). Which brunch release were you using when you had the camera working ?

saghul, 2020-06-18:
I think the camera on this MBP is connected via PCIe, but I’m not 100% sure...

sebanc, 2020-06-20:
It's interesting, I thought PCIE cameras were not working in chromeos, Do you remember which brunch release you were using when you had the camera working ?

weberk, 2021-02-13:
@saghul : Can you please show a bootlog in order to verify you have a have a usb camera? Same bootlog usually shows also the errors with its initialisation. 
I am in the same situation that my MacBook Air mid 2013 has a Facetime camera attached via pcie bus, not exposing a USB interface needed by the ChromeOS and Android Container Camera Stack.

saghul, 2021-02-13:
I'll try to remember to do this in about a week, I no longer have that laptop around.