sebanc/brunch #280: Flickering/Graphical Glitches on android apps

KaiLovesJB, 2020-06-07:
My chromeos on my surface go has been amazing and everything works perfectly except for android games, there seems to be flickering at random times and little graphical glitches like the background of the healthbar etc would disappear for a split second and then come back. This happens in every game. I am using chromeos 83 and brunch 83, the same thing happened with chromeos 81.

sebanc, 2020-06-07:
Hi,

Android apps compatibility with x86 devices is not known to be great in chromeos, especially with games.
The only thing I can think of in brunch which might cause that would be the line "--enable-hardware-overlays="single-fullscreen,single-on-top" in /etc/chrome_dev.conf.

Could you try editing the /etc/chrome_dev.conf, removing that line and checking if there is a difference ?

KaiLovesJB, 2020-06-07:
Sadly it didn’t work but thank you for answering!

closed 2020-06-26 by sebanc

ZZZXXX0110, 2021-12-01:
Despite of this issue being closed, I would like to mention that this indeed solved my exact same problem in NieR: Reincarnation in Chrome OS 94. Removing the overlay line solved all the flickering even in the "highest" graphics preset.