sebanc/brunch #281: Broadcom+Intel wifi options not working

Z3r0shin, 2020-06-07:
Hello!

I've tried the "options=broadcom_wl" and "iwlwifi_backport" on my computer that have a BCM943228HMB/QDS-BRCM1058 or a 2230BNHMW and both are not working. Either the wifi detects ssids but fail to connect to them (always says its out of range) or it simply doesn't start.

sebanc, 2020-06-07:
Hi,

Could you attch a dmesg to this issue without any "options=" ? (run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and attach dmesg.txt from your downloads folder to this issue)

lMarenda, 2020-06-08:
we had similar issues even after adding "options=broadcom_wl" to GRUB config.  fle however, that seems to resolve itself while making a second attempt to connect to Wi-Fi network.  does it get any better ?

<blockquote>

Either the wifi detects ssids but fail to connect to them (always says its out of range) or it simply doesn't start. In all case, the bluetooth works well.</blockquote>

sebanc, 2020-06-08:
@Z3r0shin Could you try the release I have just posted in "brunch-testing" repository without any option:
https://github.com/sebanc/brunch-testing/releases

Z3r0shin, 2020-06-08:
@lMarenda it seems you are right about broadcom's drivers needing 2 reboots to work right. Both machines works with the broadcom wifi chip now.

@sebanc , it seems the link you provided doesn't work. I don't really need it anymore but wanted to point it out to you anyway. Thanks again for your hard work!
Bon courage et continue ton bon travail, c'est beaucoup plus apprécié que tu peux le penser!

Z3r0shin, 2020-06-09:
The intel drivers still don't work though, even though I reset the computers more than once (which made it ok for the broadcom drivers). I don't need the intel drivers now since I can manage with the broadcom wifi chips I already have.
I will close this issue. If you want more details about my experience, feel free to reopen the issue!

Z3r0shin, 2020-06-09:
Scratch that, just got my first problems with it. Had to connect to internet through my phone. Tried wifi tethering but it didn't work. Then plugged in usb for tethering and it worked. Then, I unplugged usb, closed the computer and later restarted it. Wifi didn't work, like it was before. Even with 2-3 reboots it stayed the same. Will try using your testing branch to see if it changes anything.