sebanc/brunch #283: can chrome OS be installed together with windows 7?

djgerman10, 2020-06-07:
### can chrome OS be installed together with windows 7?
I have widows 7 already installed on the disk, but I have another clean partition, could the chrome OS be installed?

sebanc, 2020-06-08:
Hi,

If I am not mistaken, windows 7 does not have WSL so it is not possible to create a bootable chromeos image from there.

However, you can create a bootable usb or a dual boot from a linux live usb flashdrive.

djgerman10, 2020-06-08:
so it is not possible to do a dual boot with windows 7

sebanc, 2020-06-12:
The only issue is that you cannot create a chromeos usb flashdrive needed for dual boot procedure from windows 7 (as it does not have WSL).

If you already have a chromeos usb flashdrive, you can dual boot with windows 7 following the exact same procedure.

tornadox, 2020-06-12:
@djgerman10 chrome os creates lots of partitions and might break your win 7. In case you have important data I would suggest to use a separate ssd or hdddrive from chrome os, that's achievable in older laptops and in any PC or just make a USB drive in the **Install ChromeOS on a USB flash drive / SD card** wiki. If you windows 7 is MBR based probably you would need to use GRUB as the main loader for win 7 and chrome os, in win10 it's possible to use UEFI where in bios you can select the OS to boot.

closed 2020-06-28 by sebanc