sebanc/brunch #284: ThinkPad T460s. Headphone issue.

Tash254-ke, 2020-06-08:
@sebanc Thank you very much for giving me something to do and some form of escapism in these Miss Rona times.
Now, I'm a complete newbie literally the first commands I've ever written are the ones on this guide. I didn't even know how to look for HDD device name.
I have just (it's been an hour) installed chrome os fully as my main os on T460s with no issues expect that the headphone jack seems not to be working. Bluetooth audio works, I can switch between internal speakers and Bluetooth speaker seamlessly. But when I fix the earphones in, no sound and also when I select the internal speakers no audio.
Hope to hear from you soon. Greetings from 🇰🇪

sebanc, 2020-06-08:
Hi,

I am not so good at debugging audio issues and for now only rely on linux resources as chromeos and linux appear to be very similar.

Could you try to add "snd_hda_intel.model=tpt460" to the kernel command line ?

Found here: https://wiki.archlinux.org/index.php/Lenovo_ThinkPad_T460s#Sound

Tash254-ke, 2020-06-08:
[image]
I followed the procedure on this screenshot and added snd_hda_intel.model=tpt460 nothing happened when I booted.
Is this what I was supposed to do?

sebanc, 2020-06-08:
In theory it works but something must have gone wrong if it did not boot.

If you are booting from a usb flashdrive or complete hdd install, run "sudo edit-grub-config" and add "snd_hda_intel.model=tpt460" after cros_debug.

If you used the dual boot procedure, "snd_hda_intel.model=tpt460" after cros_debug in your linux distro grub config for chromeos.

Tash254-ke, 2020-06-08:
It's complete HDD install. Where will I run that sudo edit...command?
Please don't mind my questions🙈

sebanc, 2020-06-08:
No worries.

Press CTRL+ALT+T in chromeos, it will display crosh.
Type "shell" at the invite.

Then apply the above procedure.

Tash254-ke, 2020-06-08:
I did exactly as you told me and referring to the article that I took a screenshot of earlier, I spaced after cros_debug then I added snd_hda... pressed ctrl+x then typed y.
Tried playing audio via headphones immediately and there was no success so I restarted. This time it worked, problem is, once i disconnected the headphones, audio didn't play again and as long as the headphones are plugged in, internal speakers don't work either.
Turns out it only works if I start the laptop with the headphones plugged in. Once I pull them out or I try to change audio to internal speakers, it doesn't work anymore.
I've tested bluetooth too and this one works flawlessly.

Tash254-ke, 2020-06-11:
> In theory it works but something must have gone wrong if it did not boot.
>
> If you are booting from a usb flashdrive or complete hdd install, run "sudo edit-grub-config" and add [...]

I used this process to add snd_hda_intel.model=headset-mic then ctrl+x then y.
But issue still remains the same. Headphones work only when connected when booting.

sebanc, 2020-06-12:
Could you boot again with "snd_hda_intel.model=headset-mic" and post the output of "cat /sys/module/snd_hda_intel/parameters/model" ?

Tash254-ke, 2020-06-12:
You lost me at post the output of...
What's that and how do I do it?

sebanc, 2020-06-12:
Open crosh (CTRL+ALT+T), type "shell" at the invite and then type "cat /sys/module/snd_hda_intel/parameters/model"

Tash254-ke, 2020-06-13:
[image]
Followed the sudo edit-grub-config. Then added snd_hda_intel... Pressed Ctrl+x then y then enter. Restarted, then posted the output of cat/sys/module/snd_hda_intel/parameters/model

sebanc, 2020-06-13:
Could you try again with a space between "cat" and "/sys/...":
`cat /sys/module/snd_hda_intel/parameters/model`

xpduyson, 2020-06-13:
> Could you try adding "snd_hda_intel.model=headset-mic" to the kernel command line ?
> 
> Source: https://unix.stackexchange.com/questions/405640/no-headphones-output-on-debian-buster

Can confirm this fix my headphone jack issue. My laptop is Dell Latitude E5470.
Output of the above command:
```
chronos@localhost / $ cat /sys/module/snd_hda_intel/parameters/model
headset-mic,(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null)
```
[image]

Tash254-ke, 2020-06-13:
> Can confirm this fix my headphone jack issue. My laptop is Dell Latitude E5470.
> Output of the above command:
>
> ```
> chronos@localhost / $ cat /sys/module/snd_hda_intel/parameters/model [...]

The result on crosh is same as this one and now headset works anytime. However, I can't switch between internal speaker and headset when they're plugged in. But I can switch between bluetooth speaker and headset flawlessly.
As long as headset works, I'm okay.
Thank you very much. Be blessed.

closed 2020-06-26 by sebanc