sebanc/brunch #285: Touch pad, Boot, Slow computer

SPIDER-MAN859, 2020-06-08:
Touchpad not working properly
Boot takes too much time
4GB RAM but very slow
HP Notebook

sebanc, 2020-06-08:
Did you install it or are you running it from usb ? If you run it from usb it only goes as fast as your usb flahdrive.

For the touchpad, have you tried adding "options=alt_touchpad_config" to the kernel command line ?

For the boot time, yes it is slow but it is what's needed to boot make a chromeos widely compatible.

SPIDER-MAN859, 2020-06-08:
Where should I give this commands because I can't find Linux beta option
 'options=alt_touchpad_config'

SPIDER-MAN859, 2020-06-08:
And I can find only EFI Boot UEFIis not available

SPIDER-MAN859, 2020-06-08:
I have installed it on my harddrive

sebanc, 2020-06-08:
Run "sudo edit-grub-config" to change the kernel command line. Fyi in chromeos EFI partition is number 12.

Also, which recovery image have you used ?

SPIDER-MAN859, 2020-06-08:
Rammus
Processors-Intel CORE i3 7th GEN
Integrated Graphics-Intel HD Graphics 620

SPIDER-MAN859, 2020-06-08:
I am using a single boot Chrome OS

sebanc, 2020-06-08:
There is no mute button in chromeos so it is normal your mute buttons does nothing.

HP does not provide chromeos compliant bios/drivers updates so it is not possible to update. Use dual boot with windows if you need to update the bios.

There is no restart button directly accessible in chromeos but you can the hold power button and a menu with reboot should appear.

rammus recovery image has camera issues with some devices that I have not been able to fix. Try "reef" recovery image instead.

I will follow-up the issues related to your device here, could you close all the issues you have created ? (except family link which might be something more general and can remain in its own issue)

closed 2020-06-26 by sebanc