sebanc/brunch #293: Dell 7240 get's very hot after wake up on r83 k4.19 testing 20200528

kaned84, 2020-06-08:
Hi,
With brunch  r83 k4.19 testing 20200528 after wake up m dell 7240 gets very hot (like 85 celcius degree), Before going to sleep, or after reboot temperature is like 55 celcius degree max.
Laptop is dell latitude 7240 with i7-4600U processor and 16 GB ram, separate SSD 128GB drive for chrome os.

chrome os from rammus recovery image, at this point the latest stable chrome os release 83

If you need more invo let me know.

cbl5257, 2020-06-09:
I have a Latitude E5470 doing same.  On the latest beta v84.  Found that the linux vm was running cpu nearly 100%.  Ended in task mgr, load dropped and machine cooled right down.  I'm using Cog as a hardware monitor.

chunkyen, 2020-06-10:
I have 7240 with the same CPU, same brunch and same Chrome OS version 83.
No heat issue here on my side though after wake, the fan hardly spin after wake from sleep,

sebanc, 2020-06-13:
Hi all,

Could you try the latest unstable build and confirm if you still have this issue:
https://github.com/sebanc/brunch-testing/releases

You need to validate the access to the repository first in an email you have received from github.

chunkyen, 2020-06-13:
Thanks, the bonus is hyperthreading everywhere.
FYI, this is not happening to me all the time after wake from sleep in the current 20200528 brunch on Chrome OS 83. Perhaps, it is really something running at the background.
But will still try out on latest testing brunch

kaned84, 2020-06-14:
I'm going to install it today and let you know the result

chunkyen, 2020-06-14:
I don't know what you did sebanc, but the latest r83 k4.19 unstable 20200612 is working very well on Dell 7240 now, no more run away CPU temp after wake from sleep.
thanks for your good work

kaned84, 2020-06-14:
on my 7240 it does not work... gets up to 84 celcius degree, it's better - before update temp could get up to 88 celcius degree. What I ave noticed, this issue apears when you wake up laptop or log on to other account.
[image]

chunkyen, 2020-06-15:
I enabled hyperthreading by entering this kernel command "enforce_hyperthreading=1". Maybe you want to give it a try.
Also my BIOS is A25, not sure if that makes a diff.

kaned84, 2020-06-15:
should be ok, after enabling hyperthreading it's not overheating, I see 50 celcius degree max, thanks :)

cbl5257, 2020-06-18:
Working great after updating to testing branch and latest 84 beta ver

closed 2020-06-26 by sebanc