sebanc/brunch #295: Wifi and Bluetooth not work on Macbook air 2011

kelvin701, 2020-06-10:
after install chrome os, driver not support macbook.

raytrix71, 2020-06-10:
Did you try « options=broadcom_wl »?

https://youtu.be/db4Xo_xPTR4

Go to 1.01 it will show you how to add options 

( the guy in in the video is just showing how to turn on auto updates, you just have to replace the text with Broadcom)

kelvin701, 2020-06-15:
> Did you try « options=broadcom_wl »?
> 
> https://youtu.be/db4Xo_xPTR4
> 
> Go to 1.01 it will show you how to add options
> 
> ( the guy in in the video is just showing how to turn on auto updates, you just have to replace the text with Broadcom)

Oh, thank you so much, all work now after change wifi driver.

closed 2020-06-26 by sebanc