sebanc/brunch #299: Tablet Mode Issue

dkay0670, 2020-06-10:
Hello I have a Lenovo YOGA 720-15IKB 2-in-1 laptop. When I fold the laptop the keyboard disables, but tablet mode doesn't get activated. Is there a way to fix this?

tornadox, 2020-06-11:
Have you tried adding the this argument options=force_tablet_mode I got the same issue on my HP detachable laptop and this fixed it.

dkay0670, 2020-06-11:
@tornadox Well wouldn't that mean it's always in tablet mode?

dkay0670, 2020-06-11:
Nope, all that did was make it always in tablet mode

tornadox, 2020-06-11:
In that case you probably need to provide your keyboard ids, take a look at this ticker for reference 
https://github.com/sebanc/brunch/issues/186#issuecomment-632991951

dkay0670, 2020-06-11:
Okay, will do

dkay0670, 2020-06-11:
Here you go :)
`[    0.426518] input: Lid Switch as /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:18/PNP0C0D:00/input/input0
[    0.426581] input: Power Button as /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0C:00/input/input1
[    0.426636] input: Power Button as /devices/LNXSYSTM:00/LNXPWRBN:00/input/input2
[    0.459812] input: AT Translated Set 2 keyboard as /devices/platform/i8042/serio0/input/input3
[    1.712491] input: EasyCamera: EasyCamera as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/input/input4
[   19.164636] input: Video Bus as /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/LNXVIDEO:00/input/input5
[   20.753470] input: Tablet mode switch as /devices/platform/tablet_mode_switch.0/input/input6
[   20.771867] logitech-djreceiver 0003:046D:C52B.0003: hiddev0,hidraw0: USB HID v1.11 Device [Logitech USB Receiver] on usb-0000:00:14.0-2/input2
[   20.831261] snd_hda_codec_realtek hdaudioC0D0:    inputs:
[   20.874283] input: HDA Intel PCH Mic as /devices/pci0000:00/0000:00:1f.3/sound/card0/input7
[   20.875186] input: HDA Intel PCH Headphone as /devices/pci0000:00/0000:00:1f.3/sound/card0/input8
[   20.875251] input: HDA Intel PCH HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input9
[   20.875322] input: HDA Intel PCH HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input10
[   20.875379] input: HDA Intel PCH HDMI/DP,pcm=8 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input11
[   20.875467] input: HDA Intel PCH HDMI/DP,pcm=9 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input12
[   20.875563] input: HDA Intel PCH HDMI/DP,pcm=10 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input13
[   20.877980] input: Logitech Unifying Device. Wireless PID:405e Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.2/0003:046D:C52B.0003/0003:046D:405E.0004/input/input14
[   20.878160] input: Logitech Unifying Device. Wireless PID:405e Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.2/0003:046D:C52B.0003/0003:046D:405E.0004/input/input15
[   20.878264] input: Logitech Unifying Device. Wireless PID:405e Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.2/0003:046D:C52B.0003/0003:046D:405E.0004/input/input16
[   20.878330] input: Logitech Unifying Device. Wireless PID:405e System Control as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.2/0003:046D:C52B.0003/0003:046D:405E.0004/input/input17
[   20.879354] hid-generic 0003:046D:405E.0004: input,hidraw1: USB HID v1.11 Keyboard [Logitech Unifying Device. Wireless PID:405e] on usb-0000:00:14.0-2:1
[   20.934240] input: WCOM50FC:00 056A:50FC Touchscreen as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-WCOM50FC:00/0018:056A:50FC.0006/input/input21
[   20.934598] input: WCOM50FC:00 056A:50FC as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-WCOM50FC:00/0018:056A:50FC.0006/input/input22
[   20.950628] input: WCOM50FC:00 056A:50FC as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-WCOM50FC:00/0018:056A:50FC.0006/input/input23
[   20.968548] input: WCOM50FC:00 056A:50FC as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-WCOM50FC:00/0018:056A:50FC.0006/input/input24
[   20.968640] input: WCOM50FC:00 056A:50FC Mouse as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-WCOM50FC:00/0018:056A:50FC.0006/input/input25
[   20.968728] hid-generic 0018:056A:50FC.0006: input,hidraw1: I2C HID v1.00 Mouse [WCOM50FC:00 056A:50FC] on i2c-WCOM50FC:00
[   20.975857] input: Logitech M720 Triathlon as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.2/0003:046D:C52B.0003/0003:046D:405E.0004/input/input26
[   20.976218] logitech-hidpp-device 0003:046D:405E.0004: input,hidraw3: USB HID v1.11 Keyboard [Logitech M720 Triathlon] on usb-0000:00:14.0-2:1
[   20.999777] input: ELAN0651:00 04F3:3052 Touchpad as /devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-8/i2c-ELAN0651:00/0018:04F3:3052.0007/input/input28
[   20.999882] hid-multitouch 0018:04F3:3052.0007: input,hidraw4: I2C HID v1.00 Mouse [ELAN0651:00 04F3:3052] on i2c-ELAN0651:00
[   21.085964] input: Wacom HID 50FC Pen as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-WCOM50FC:00/0018:056A:50FC.0006/input/input29
[   21.086073] input: Wacom HID 50FC Finger as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-WCOM50FC:00/0018:056A:50FC.0006/input/input30`

sebanc, 2020-06-12:
Hi,

In order for tablet mode to be detected, the "udev" program needs to receive tablet mode related events.

Could you run "sudo udevadm monitor", then switch your laptop to tablet mode and post the output here ?

dkay0670, 2020-06-12:
Nothing was outputted other then some auto brightness events, overall It's just not detecting anything useful to this issue, but I will attach it anyway.

`
monitor will print the received events for:
UDEV - the event which udev sends out after rule processing
KERNEL - the kernel uevent

KERNEL[190.885787] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
UDEV  [190.888998] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
KERNEL[190.906075] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
UDEV  [190.907321] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
KERNEL[190.926344] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
UDEV  [190.927688] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
KERNEL[190.948160] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
UDEV  [190.950896] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
KERNEL[190.968828] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
UDEV  [190.971559] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
KERNEL[190.989572] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
UDEV  [190.992170] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
KERNEL[191.010158] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
UDEV  [191.012408] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
KERNEL[191.030553] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
UDEV  [191.032646] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
KERNEL[191.051127] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
UDEV  [191.053943] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
KERNEL[191.071856] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
UDEV  [191.074818] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
KERNEL[198.637652] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
UDEV  [198.640718] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
KERNEL[198.658662] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
UDEV  [198.661178] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
KERNEL[198.678903] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
UDEV  [198.681697] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
KERNEL[198.699462] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
UDEV  [198.702221] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
KERNEL[198.721018] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
UDEV  [198.723447] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
KERNEL[198.740593] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
UDEV  [198.743457] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
KERNEL[198.761315] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
UDEV  [198.764106] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
KERNEL[198.781946] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
UDEV  [198.784647] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
KERNEL[198.802937] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
UDEV  [198.804327] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
KERNEL[198.823443] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
UDEV  [198.825226] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
KERNEL[199.029438] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
UDEV  [199.032736] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
KERNEL[199.050195] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
UDEV  [199.053084] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
KERNEL[199.071542] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
UDEV  [199.073831] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
[... cut ...]
UDEV  [217.133177] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
KERNEL[217.151745] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
UDEV  [217.154099] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
KERNEL[217.172370] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
UDEV  [217.174479] change   /devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight (backlight)
`

EETagent, 2020-06-12:
Hello, I would like to ask here about tablet mode on Acer Spin 1 instead of opening new issue. Keyboard is disabled but ChromeOS does not enter tablet mode.
`sudo udevadm monitor` Doesn't output anything when fliping screen.  

Thanks for your work, everything except accelerometer and stable HDMI output works absolutely great.

dkay0670, 2020-06-12:
@EETagent I think personally you should open a new issue, it might get a bit confusing.

dkay0670, 2020-06-12:
I'm thinking Chrome OS doesn't see it, but the hardware itself is disabling the keyboard.

EETagent, 2020-06-12:
@dkay0670  Probably not. If this is true, keyboard would be disabled even at boot. That is not happening.

dkay0670, 2020-06-12:
Hmmm, okay

sebanc, 2020-06-13:
Thanks for the udev log.

> I'm thinking Chrome OS doesn't see it, but the hardware itself is disabling the keyboard.

Indeed that's exactly it, the keyboard is disabled by the hardware (EC) when you fold the laptop but "TABLET_MODE" event is not sent and there is no real keyboard disconnection event that could be used to detect this.

The main problem is that the Linux kernel lacks a lot of tablet mode detection drivers as Linux DE don't really use it.

Here are the possibilities from here:

1) Try to look around on Linux forums / GitHub and see if there is not someone nice who wrote a 
 kernel module for tablet mode detection on your device. If you point me to it, I can try to implement it. (it should be kernel code, not weird bash script that hijack acpi events).

2) Use the keyboard shortcut CTRL+ALT+SHIFT+T to toggle manually between laptop mode and tablet mode (this shortcut should work for most but might not work if you have a kernel driver reporting tablet mode capability even though it does not actually send tablet mode events).

3) Use "options=sysfs_tablet_mode" and trigger tablet mode from crosh with:
`echo 1 | sudo tee /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode`

Alternatively, if you want your device to always be in tablet mode, use "options=force_tablet_mode"

EETagent, 2020-06-13:
@sebanc Thanks. CTRL+ALT+SHIFT+T did the trick. I do not longer need automatic tablet mode switch.

sebanc, 2020-06-13:
Yes it is because chromeos considers the physical keyboard is still enabled. Using "options=sysfs_tablet_mode" is a bit more complicated but chromeos should properly disable the physical keyboard and show the on-screen one.

dkay0670, 2020-06-13:
Okay, thanks for the helo

dkay0670, 2020-06-13:
I will reopen this issue if I find the driver

closed 2020-06-13 by dkay0670