sebanc/brunch #302: ChromeOS on VMWare and Virtualbox

vineethrp, 2020-06-11:
Now that we have chromeos booting on most x86 hardware, would it be possible to enable support for virtualization platforms like vmware, virtualbox etc.

I did give it a try once, but did not boot. Will try again and post the logs here

vineethrp, 2020-06-11:
[image]

princepjamie06832, 2020-06-12:
It won't run on virtual machine. Only be installed on your computer.

sebanc, 2020-06-12:
Hi,

I am closing this issue with regards to the explanations provided in issue #303. To have chromeos working in Virtualbox, you have first to find a way to add virtualbox mesa drivers to chromeos.

closed 2020-06-12 by sebanc

pagkly, 2020-06-17:
> Now that we have chromeos booting on most x86 hardware, would it be possible to enable support for virtualization platforms like vmware, virtualbox etc.
> 
> I did give it a try once, but did not boot. Will try again and post the logs here

mine has similar problem but it stuck at
[ 51.782565] input: ImExPS/2 Generic Explorer Mouse as devices/platform/i8042/serio1/input/input6

szalony9szymek, 2020-11-20:
for me it gets stuck I allowed UEFI

dragonBall561, 2021-03-02:
what I did w/ VMware

1. SATA Hard Drive
2. No VT-X
3. UEFI Boot Mode

Result. So close...
https://drive.google.com/file/d/14xkdi3vKAS-henCKulbuIbkEF1kFILLH/view?usp=sharing

Warrentheo, 2021-12-20:
A distant relative to this issue, but you might want to check this out:
https://github.com/WSA-Community/WSAGAScript
_Working for me_