sebanc/brunch #303: Booting ChromeOS Intel recovery images with Nvidia / AMD / Vmware graphics only (no i915)

sebanc, 2020-06-12:
Currently, ChromeOS will not boot when there is no Intel GPU because only i915 graphics userspace drivers are included in the recovery images (you can easily confirm it by running "ls /usr/lib64/dri/").

During the last 2 days I have tried to build from source and add the full "libdrm" and "mesa" drivers to ChromeOS, however it did not succeed and I have no more time to spend on this.

If someone finds a proper way to add other userspace graphic drivers to chromeos recovery images, I don't mind to implement it but until then:
- Issues related to booting Intel recovery images without i915 (only Nvidia / AMD GPU) will be instantly closed.
- Issues related to ChromeOS not booting under Virtualbox (Vmware GPU) will also be instantly closed.

TipzTeam, 2020-06-15:
Hmm, interesting, I would try to get it booted on my amd or vbox, if I ever do I will post instructions here.

dragonBall561, 2021-03-19:
If on VMWare, do:

1. Hard Drive Type=sata
2. UEFI boot=on

tejasraman, 2022-10-21:
FYI Ryzen and StoneyRidge/BristolRidge iGPUs work too.

Jhone-Harry, 2022-11-13:
Hello, Installing ChromeOS on VMware Workstation Pro is what I want to do. There is a statement that "visual machines are not supported" on the GitHub downloads page. Is it possible to make work on VMware Workstation or VirtualBox?
Please, if you can assist, I appreciate it.

tejasraman, 2022-11-13:
No - because the VMware GPU doesn't work with the chromeOS drivers.

Trust me, I tried (with Workstation Player 16).

Jhone-Harry, 2022-11-14:
QEMU may be used to install Chrome OS, I believe.
@tejasraman

tejasraman, 2022-11-14:
Yes - the QEMU/KVM engine can emulate chipsets so you can get an unaccelerated VM there. It'll be slow though 🐌

xAffan, 2023-06-24:
Can't KVM do passthrough or something to achieve hardware acceleration?

tejasraman, 2023-06-24:
> Can't KVM do passthrough or something to achieve hardware acceleration?

Yes, if you can pass through your iGPU you'd be good

zebn0710, 2023-07-06:
cna anyone send me there exported ova file i cant even get it to boot

tejasraman, 2023-07-06:
> cna anyone send me there exported ova file i cant even get it to boot 
> 

No one has an exported OVA file because unless you set up PCIe passthrough with KVM on Linux, you will not get it. (To do that you need an iGPU and a dGPU.

LagLifeYT, 2024-02-20:
So, I've been working on this for a while. Running ChromeOS in VMware requires vmwgfx (VMware SVGA 3D kernel module) built into the kernel itself. This is actually quite similar to how this works on a real Chromebook, there's no i915 module visible when running lsmod on my Pixelbook Go. 

After booting with a kernel that has VMware's graphics drivers in the kernel (and not a module), the ChromeOS logo appears and then it tries to launch Chrome. Frecon seems to work at least since the bootsplash appears to depend on it to show up. There's clearly some graphics library issue since under FydeOS for VMware replacing the Chromium binary with one from a very close version of ChromeOS works just fine. Further experimentation is needed to figure out _what_ is broken. I suspect that introducing new graphics drivers for other platforms would work very much the same way once we figure it out.

Forge64, 2024-02-27:
I know vmwgfx is probably right out, but has this changed for amdgpu, with AMD APUs being in "real" Chromebooks now? What's the current DRM status in the brunch kernel? i915? radeon? amdgpu? Do we have Intel ARC/XE support now?

LagLifeYT, 2024-03-10:
> I know vmwgfx is probably right out, but has this changed for amdgpu, with AMD APUs being in "real" Chromebooks now? What's the current DRM status in the brunch kernel? i915? radeon? amdgpu? Do we have Intel ARC/XE support now?

DRM for all platforms are modules for some reason. Kind of silly since I believe Linux will choose the correct DRM driver when it finds a platform it has the driver for.