sebanc/brunch #305: Rules to report touchpad issues

sebanc, 2020-06-12:
1) Check if your touchpad driver is correctly detected or not by installing crouton ("easy way" configuration) and see if the touchpad works there or not.

If it does not work, you can create an issue, else continue reading.

2) If your touchpad works in crouton but not in chromeos, it is due to a specific configuration needed by the cmt driver used in chromeos.

In that case, you can try to modify the first section of the file /etc/gesture/40-touchpad-cmt.conf to better manage your touchpad using other options found in that same file or the ones available at:
https://github.com/hugegreenbug/xf86-input-cmt/tree/master/xorg-conf

If you find a config that works and you want it to be present in brunch as an option, do not hesitate to create an issue with the related config.

3) I tried many configs but nothing worked.

Unfortunately, there is not much I can do in that case as I do not even have access to the related hardware.

jafer241177, 2020-08-28:
Why doesn't a tool like Linux take drivers and hardware updates automatically without writing it

jafer241177, 2020-08-28:
Is it possible to change the BIOS drivers so that it becomes like chrome and automatically recognizes hardware updates?

jafer241177, 2020-08-29:
P: /devices/pci0000:00/0000:00:16.3/i2c_designware.1/i2c-7/i2c-ELAN238E:00/0018:04F3:2A7C.0001/input/input5/mouse0
N: input/mouse0
E: DEVNAME=/dev/input/mouse0
E: DEVPATH=/devices/pci0000:00/0000:00:16.3/i2c_designware.1/i2c-7/i2c-ELAN238E:00/0018:04F3:2A7C.0001/input/input5/mouse0
E: ID_INPUT=1
E: ID_INPUT_TOUCHSCREEN=1
E: ID_PATH=pci-0000:00:16.3-platform-i2c_designware.1
E: ID_PATH_TAG=pci-0000_00_16_3-platform-i2c_designware_1
E: MAJOR=13
E: MINOR=32
E: POWERD_ROLE=internal_touchscreen
E: POWERD_TAGS=usable_when_laptop usable_when_tablet role_internal_touchscreen
E: POWERD_TAGS_USABLE=usable_when_laptop usable_when_tablet
E: SUBSYSTEM=input
E: TAGS=:powerd:
E: USEC_INITIALIZED=28420035

 udevadm info /dev/input/mouse1
P: /devices/pci0000:00/0000:00:17.0/i2c_designware.2/i2c-8/i2c-SYNA2392:00/0018:06CB:7FB5.0002/input/input10/mouse1
N: input/mouse1
S: input/by-path/pci-0000:00:17.0-platform-i2c_designware.2-mouse
E: DEVLINKS=/dev/input/by-path/pci-0000:00:17.0-platform-i2c_designware.2-mouse
E: DEVNAME=/dev/input/mouse1
E: DEVPATH=/devices/pci0000:00/0000:00:17.0/i2c_designware.2/i2c-8/i2c-SYNA2392:00/0018:06CB:7FB5.0002/input/input10/mouse1
E: ID_INPUT=1
E: ID_INPUT_TOUCHPAD=1
E: ID_PATH=pci-0000:00:17.0-platform-i2c_designware.2
E: ID_PATH_TAG=pci-0000_00_17_0-platform-i2c_designware_2
E: ID_SERIAL=noserial
E: MAJOR=13
E: MINOR=33
E: POWERD_ROLE=internal_touchpad
E: POWERD_TAGS=usable_when_laptop usable_when_display_off role_internal_touchpad
E: POWERD_TAGS_USABLE=usable_when_laptop usable_when_display_off
E: SUBSYSTEM=input
E: TAGS=:powerd:
E: USEC_INITIALIZED=28427430

tray to improve my touchpad

pierro78, 2021-11-08:
2 fingers scrolling will sometimes stop working on my surface laptop go but the  "psmouse.elantech_smbus=1" kernel option seems to fix the issue 🙂