sebanc/brunch #307: Please read this before creating issues

sebanc, 2020-06-12:
Before creating an issue, first make sure that you have read the main README of this repo.

Also, it is important to consider that I am not related to any youtube video and don't provide support if you have installation issues while following those. Just follow the instructions in the main README of this repo instead.

If your issue concerns "Booting ChromeOS Intel recovery images with Nvidia / AMD graphics only (no intel gpu) or in VirtualBox", please read #303.

If your issue concerns "ChromeOS startup being too long with brunch". please read #304.

If your issue concerns "Touchpad support", please read #305.

If your issue concerns "Auto-rotation / Ambient light sensor not working", please read #306.

TipzTeam, 2020-06-15:
Oh thank you and you should add Bluetooth stuff like how to enable it by themselves

dkay0670, 2020-06-22:
@sebanc Discord Invite invalid

ameyjoshi3068, 2022-10-15:
I think we should also add this
 updating to ChromeOS to 103 or higher is not suggested. If you want to use ChromeOS 103, please use Brunch 103 (or higher) and perform a fresh installation. 

Additionally: There are many known issues with recent releases of ChromeOS, most of which are caused by bugs in the OS itself and cannot be fixed by brunch. If you experience any issues that are not covered on the Github's Troubleshooting and Support page, try to reinstall using ChromeOS 94 instead. This version is generally considered the most stable release. You will need to use Brunch 102 for ChromeOS 102 and older, or Brunch 103+ for ChromeOS 103 and higher. You can find old releases here

Rammus 94:
https://dl.google.com/dl/edgedl/chromeos/recovery/chromeos_14150.74.0_rammus_recovery_stable-channel_mp-v2.bin.zip

Volteer 94:
https://dl.google.com/dl/edgedl/chromeos/recovery/chromeos_14150.87.0_volteer_recovery_stable-channel_mp-v8.bin.zip

Grunt 94:
https://dl.google.com/dl/edgedl/chromeos/recovery/chromeos_14150.87.0_grunt_recovery_stable-channel_mp-v6.bin.zip

Zork 94:
https://dl.google.com/dl/edgedl/chromeos/recovery/chromeos_14150.87.0_zork_recovery_stable-channel_mp-v7.bin.zip

Brunch 102 should be used with all ChromeOS releases before 103.
Brunch 102:
https://github.com/sebanc/brunch/releases/download/r102-stable-20220625/brunch_r102_stable_20220625.tar.gz

Rstment25, 2023-02-18:
> I think we should also add this updating to ChromeOS to 103 or higher is not suggested. If you want to use ChromeOS 103, please use Brunch 103 (or higher) and perform a fresh installation. [...]
 
Definitely should be added... I had no internet access in setup after installing chromeos 109 - and it never told me that, just white screen with nothing going on ( No error message informing me at all of what's going on... )

If not for luck that I thought of searching in this direction I would have never figured it out... Most of the issues on github have the error message informing them that the internet is broken ,but I guess I'm amongst those who don't... who would even search for "No Internet" when all you see is white screen during setup 

No doubt it would help a LOT of users if @sebanc simply added at the end of install instructions : If you're facing issues download chromeos 94 and brunch 102