sebanc/brunch #308: Please read before reporting Tablet Mode detection issues

sebanc, 2020-06-13:
Currently, the Linux kernel lacks a lot of tablet mode detection drivers (most Linux DE don't use it).

Here are the possibilities you have if Tablet Mode detection is not working on your device:

1) Run "sudo udevadm monitor" in crosh shell and fold your device in tablet mode. If you see keyboard disconnection events, tablet mode switching can probably be reproduced based on those events. In that case, create an issue and attach "sudo udevadm monitor" output.

2) Try to look around on Linux forums / GitHub and see if there is not someone nice who wrote a
kernel module for tablet mode detection on your device. If that's the case, create an issue with the link to the driver and I will try to implement it. (it has to be kernel code, not bash scripts that hijack acpi events).

3) Use the keyboard shortcut CTRL+ALT+SHIFT+T to toggle manually between laptop mode and tablet mode (this shortcut might not work if you have a kernel driver reporting tablet mode capability even though it does not actually send tablet mode events).

4) Use "options=sysfs_tablet_mode" and trigger tablet mode from crosh with:
echo 1 | sudo tee /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode

Alternatively, if you want your device to always be in tablet mode, use "options=force_tablet_mode"

Skawtnyc, 2020-06-13:
I chose to use force_tablet_mode on my Surface Go. It works, and when I have it connected to my desktop setup with keyboard and mouse, it properly shifts to desktop mode.

CarlitoisLoco, 2020-08-02:
Is there a way to force clamshell Mode? My laptop has been forced to tablet mode at all times and ctrl+alt+shift+t doesn't work

JosephInn, 2020-09-26:
Unfortunately, Tablet Mode (Manual Triggering) is Not Working for me. Here's all the info:
[Description&Logs](https://docs.google.com/document/d/1TbYgcp4W0jKRHjsXvTw3OQ2gR-wLRZjUb28L6TgSnJk/edit?usp=sharing)