sebanc/brunch #309: Sound on HP Pro 608 G1

kbnhvn, 2020-06-13:
Sound is not working on this tablet, soundcard is bytcht-cx2072x.
I've read [https://github.com/sebanc/brunch/issues/85](url) and tested files from [https://github.com/heikomat/linux/tree/cx2072x/cx2072x_fixes_and_manual](url) and from [https://github.com/alsa-project/alsa-ucm-conf/archive/master.zip](url) but still not working.

aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Audio [Intel HDMI/DP LPE Audio], device 0: HdmiLpeAudio [Intel HDMI/DP LPE Audi]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: Audio [Intel HDMI/DP LPE Audio], device 1: HdmiLpeAudio [Intel HDMI/DP LPE Audi]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: Audio [Intel HDMI/DP LPE Audio], device 2: HdmiLpeAudio [Intel HDMI/DP LPE Audi]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: bytchtcx2072x [bytcht-cx2072x], device 0: Audio (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: bytchtcx2072x [bytcht-cx2072x], device 1: Deep-Buffer Audio (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

sebanc, 2020-06-14:
Hi, could you try the new testing release ?

kbnhvn, 2020-06-14:
Sound works great ! Many thanks !!

closed 2020-06-26 by sebanc