sebanc/brunch #310: Linux Support Not Working

dkay0670, 2020-06-13:
I have installed brunch, got everything but the Linux Support (Beta) feature working. I'm making this issue because **I know for a fact** that it can work, as I got it to work once before I reinstalled brunch. It downloads and installs fine, but won't start the container.

Here is a screenshot of the error:
[image]

sebanc, 2020-06-14:
Hi,

Which brunch release do you use ?
What are your laptop specs ?
Are you running chromeos on hdd or usb ?

dkay0670, 2020-06-14:
Brunch 83.
It's running on the internal HDD.
Intel i7-7700HQ (4) @ 3.800GHz

sebanc, 2020-06-14:
That's very weird, it should work with your hardware. Could you try to remove crostini in settings and to reinstall it.

If it still does not work, could you try the new build ?

dkay0670, 2020-06-14:
I can do that

dkay0670, 2020-06-14:
Although I have tried reinstalling crostini

dkay0670, 2020-06-15:
That worked, thanks!

dkay0670, 2020-06-18:
It's not working again...

closed 2020-06-18 by dkay0670

MilesPong, 2020-07-04:
I'm also having this same issue, anyone solved it at last?

My device info,

Brunch r83 k4.19 testing 20200614
HP Envy13 notebook, Intel i5-6500, 8GB RAM
Only 1 SSD, and boot from one of the partitions from SSD (The others are Window 10 and Ubuntu 20.04, total have 3 OS)   

BTW, the first boot I got a black screen but solved it by replace the grubx64.efi mentioned in #30

dkay0670, 2020-07-04:
@MilesPong I fixed it by going on my mobile hotspot

MilesPong, 2020-07-05:
Seems to update to latest brunch release 0704 has solved this issue