sebanc/brunch #311: Netflix Android App Won't Play Anything.

Tash254-ke, 2020-06-13:
[image]
I sideloaded an app from APK Mirror, logged in, downloaded some movies and series episodes. But nothing plays whether downlaoded or streaming. Is there a work around to this?
I need downloads coz I live in the country side and we don't have FTTH yet and normal data can be expensive.

Skawtnyc, 2020-06-14:
It's the DRM in the app. You will only be able to use the website version of Netflix using Brunch.

raytrix71, 2020-06-14:
It’s currently working for me

sebanc, 2020-06-14:
Netflix needs widevine drm which is only available after disabling developer mode (by removing "cros_debug" on the kernel command line) and doing a powerwash.

However, note that disabling developer mode will restore selinux which might prevent chromeos from booting or some devices (wifi notably) from functionning.

Skawtnyc, 2020-06-14:
Ah, that explains why I never had video in the Netflix app. Fortunately the website works fine, I can use the Android app on my Duet tablet, and I actually need to have dev mode for my normal workflow.

Tash254-ke, 2020-06-15:
> It’s currently working for me

The app itself? Or the website?

Tash254-ke, 2020-06-15:
> The official Netflix app (with Download capability) needs widevine drm which is only available after disabling developer mode (by removing "cros_debug" on the kernel command line) and doing a powerwash.
> 
> However, note that disabling developer mode will restore selinux which might prevent chromeos from booting or some devices (wifi notably) from functionning.

😪 That's fine I'll look for otherways. Thank you once again for Brunch.

Tash254-ke, 2020-06-15:
> Ah, that explains why I never had video in the Netflix app. Fortunately the website works fine, I can use the Android app on my Duet tablet, and I actually need to have dev mode for my normal workflow.

How long have you had the Duet? Is it a good device? Mama needs a device to just to some simple work and I had suggested this device. Problem is it's not yet here.

Skawtnyc, 2020-06-15:
> How long have you had the Duet? Is it a good device? Mama needs a device to just to some simple work and I had suggested this device. Problem is it's not yet here.

Off-topic, but I will say that it's great as a tablet. I use it all the time when I'm not at my desk.

closed 2020-06-28 by sebanc