sebanc/brunch #312: bcm4314 issues

TipzTeam, 2020-06-14:


kaned84, 2020-06-14:
did you try brunch options for wifi card?

sebanc, 2020-06-14:
Try adding "options=broadcom_wl" to the kernel command line.

TipzTeam, 2020-06-15:
Wlan will definitely be fixed by that but Bluetooth? This is a BT+WLAN card

Also you mean adding that by running like export command or the kernel makefile?

tornadox, 2020-06-15:
@TipzTeam just check the https://github.com/sebanc/brunch/wiki/Getting-Started under the Framework options. The guide is quite simple. I think the question about wifi compatibility is somewhat similar to hackintosh communities, where if it doesn't work (like an intel wifi) you need to replace the module.

TipzTeam, 2020-06-17:
> @TipzTeam just check the https://github.com/sebanc/brunch/wiki/Getting-Started under the Framework options. The guide is quite simple. I think the question about wifi compatibility is somewhat [...]

You mean kexts?

sebanc, 2020-06-29:
I guess this issue is fixed, if not feel free to re-open it

closed 2020-06-29 by sebanc

TipzTeam, 2020-07-14:
> I guess this issue is fixed, if not feel free to re-open it

I am unable to compile myself but it is possible to compile correctly, idk why it doesn't work for me.