sebanc/brunch #313: Install fails on resize2fs

pheppy, 2020-06-14:
During install I get a "resize2fs" error (see image) 
[image]
and thereafter many "no diskspace left on device" errors.
I've installed via SD, via Mint liveCD. Same result. Previous versions of chromeos and brunch.
I'm using the MBR version which seems to work.

sebanc, 2020-06-14:
Hi,

The issue is not resize2fs but "ILL_ILLOPN" and "trap invalid opcode" errors which means that the recovery image uses cpu instructions not supported by your cpu.

What are your laptop specs ?

pheppy, 2020-06-14:
Old. ASUS UL50V with an Intel U7300 processor, so I used samus. It works on cloudready, so I thought it had a chance.

sebanc, 2020-06-14:
No, sorry there are no current chromeos recovery images compatible with core 2 duo. Cloudready works because it is built from source without specific cpu instructions, whereas google includes specific cpu instructions in recovery images to improve devices performance.

pheppy, 2020-06-14:
OK. maybe sometime.

closed 2020-06-14 by pheppy