sebanc/brunch #317: Dell 7240 keyboard backlight problem - does not remeber setup

kaned84, 2020-06-15:
Hi there,

I have this problem since the begining of using Chrome OS on my laptop. It just does not remember backlight settings, after booting it always sets backlight on. when I turn it off after some time it is turned on again. Also sometimes like watching youtube or netflix it turns on and off spontaneusly. Currently my dell is on latest testing brunch  20200614 and on chrome os stable 83.0.4103.97.

PS. if it is usefull info - I have a touchscreen in it.

Pumpino, 2020-06-16:
I have the same issue on my 7250. I prefer having the keyboard backlight disabled (to conserve battery), and it re-enables itself.

au79912, 2020-06-16:
I had the same issue on my dell E6440, i dont think you can do anything about the backlight turn on when it first boots ecpt disable it in the bios, as for the random turning on and off again you can try using the chrome os specific shorcut to control backlight (alt+F6(to turn down)) instead of the dells build in  Fn shortcut. that solved the problem for me

Pumpino, 2020-06-16:
The keyboard backlight is disabled already in my bios. I'm running the latest bios (released this week).

I think I'll have to use CloudReady on mine (the issue isn't present) and stick to Brunch on my HP ProDesk 400 (everything works fine on it).

kaned84, 2020-06-16:
tried tip from au79912 and I think it works, thanks! :) alt+F6(-) /alt+F7(+)

sebanc, 2020-06-16:
As for screen brightness, chromeos does not store keyboard backlight brightness level and reset it to the chromeos default value on reboot. That's a standard chromeos behavior.

closed 2020-06-26 by sebanc