sebanc/brunch #319: Touchscreen mapped to wrong monitor in multimonitor

kowalski7cc, 2020-06-16:
System:
XPS 13 9350 with 4K screen
HDMI external screen 1080P

Issue:
When the laptop is connected to the second screen, the touch is mapped to the external monitor instead of the internal one.

sebanc, 2020-06-16:
Hi, is your main laptop screen set as primary in chromeos ?

kowalski7cc, 2020-06-16:
Yes it is. If I set the external one as primary, the mapping is still wrong.

kowalski7cc, 2020-06-17:
Forgot to mention nami recovery, but it happens with every chrome/chromium os build. It does work fine on Fedora

sebanc, 2020-06-29:
Hi,

Just an update to say that for now I have no idea why the touchscreen mapping is incorrect when an external monitor is present. My understanding is that the mapping of the touchscreen and the display is performed by chromeos (not the kernel) and I did not find specific config files / settings related to this.

Nevertheless, I just noticed the chrome flag #enable-touchscreen-calibration, could you try to see if it might help ?

daemongloom, 2020-06-30:
> Nevertheless, I just noticed the chrome flag #enable-touchscreen-calibration, could you try to see if it might help ?

That flag does not add any options neither to specified url (chrome://settings/display) nor to settings app.
Issue happens with both displaylink and usb-c(alternate mode) external displays
Touchscreen info for my device (HP Spectre X2 2017) below:
`[    3.081393] usb 1-9: Manufacturer: ELAN
[    3.095030] input: ELAN Touchscreen Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0/0003:04F3:2221.0002/input/input9
[    3.098810] input: ELAN Touchscreen as /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0/0003:04F3:2221.0002/input/input10
[    3.102655] hid-multitouch 0003:04F3:2221.0002: input,hiddev97,hidraw1: USB HID v1.10 Device [ELAN Touchscreen] on usb-0000:00:14.0-9/input0`
`Input device ID: bus 0x3 vendor 0x4f3 product 0x2221 version 0x110
Input device name: "ELAN Touchscreen"`
[lsusb -v](https://github.com/sebanc/brunch/files/4850498/elan_touchscreen_info.txt)

kowalski7cc, 2020-06-30:
I have the same touchscreen brand:
Port 4: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        ID 04f3:20d0 Elan Microelectronics Corp.

`Bus 001 Device 003: ID 04f3:20d0 Elan Microelectronics Corp. Touchscreen
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x04f3 Elan Microelectronics Corp.
  idProduct          0x20d0 
  bcdDevice           11.12
  iManufacturer           4 ELAN
  iProduct               14 Touchscreen
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     945
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               2
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered`

kowalski7cc, 2020-06-30:
I have the same issue also on other Chromium OS builds (Arnoldthebat and CloudReady).
No issues on Fedora with Wayland. (Xorg treats multimonitor setups as one big display)

daemongloom, 2020-09-01:
I have Lenovo Duet now, so I could check if that issue exists in real Chromebook.
No, touchscreen works on correct display there. I was using the same dock/monitor for that test.

kowalski7cc, 2020-10-13:
I don't know why, on some monitors/adapters is mapped correctly, on others is not.

closed 2026-04-22 by github-actions[bot]