sebanc/brunch #320: ACPI Bios Error

desmondam, 2020-06-16:
I don't know if this trouble is caused because my pc has any issue. I am a newbie.
When I try to boot the grub boot loader appears and I select chromeos. Then the booting log moves fastly and after some time the movement of the log stops showing
ACPI Bios error....
...
...
...
AE_NOT_FOUND while evaluating GPE method
This appears four times and after that a short log is shown and the system reboots
This reoccurs like a cycle
Can anyone help me out?
I first tried chromefy and just got a blank screen on boot and it stuck there
Then I tried brunch.. but I am stuck with this problem
Please help me out of here

sebanc, 2020-06-16:
Hi, what are your laptop/pc specs ?

desmondam, 2020-06-17:
Thanks for replying sebanc
My laptop is a low end one it is a Toshiba Sattelite C50-A. It's specs are:
>Intel Celeron CPU N2820 @2.13Ghz 2.13Ghz
>2gb Ram

sebanc, 2020-06-17:
Which recovery image are you using ? Could you post a short video of the boot process ? (The ACPI warnings you mention are generally harmless and I doubt it is the real issue)

desmondam, 2020-06-17:
I also doubt it because it also appears in ubuntu while booting but it does not affect booting it works perfectly
I used eve recovery image

sebanc, 2020-06-17:
Pentium CPU are not compatible with eve, try rammus recovery image

desmondam, 2020-06-17:
Sorry sebanc I used eve because I downloaded it for check chromefy.
But today I use sammus as instructed in read me.
It just worked great
But it is bit laggy.I don't know if my computer does not have the specs to run chrome os.please reply me on this.
But your work is so great.Thanks for that.You are just awesome

sebanc, 2020-06-17:
Not sure why it is laggy but you can try "rammus" or "reef" images which are better and should be compatible with your cpu

desmondam, 2020-06-17:
Ok i will try it

desmondam, 2020-06-18:
I tried to use rammus recovery
but when i run sudo bash chromeos-install.sh -src recovery.bin -dst /dev/sdc
It's giving me an error:
chromeos-install.sh: warning: chromeos-install.sh: warning: script from noexec mount; see https://chromium.googlesource.com/chromiumos/docs/+/master/security/noexec_shell_scripts.md
WARNING: Primary GPT header is invalid
WARNING: Secondary GPT header is invalid
ERROR: GptSanityCheck() returned 2: GPT_ERROR_INVALID_HEADERS
chromeos-install.sh: line 187: [: -eq: unary operator expected
WARNING: Primary GPT header is invalid
WARNING: Secondary GPT header is invalid
ERROR: GptSanityCheck() returned 2: GPT_ERROR_INVALID_HEADERS
chromeos-install.sh: line 187: [: -gt: unary operator expected
WARNING: Primary GPT header is invalid
WARNING: Secondary GPT header is invalid
ERROR: GptSanityCheck() returned 2: GPT_ERROR_INVALID_HEADERS
.....
....
....
partx: specified range <1:0> does not make sense
Writing partition 1
pv: /dev/loop5p1: No such file or directory
Writing partition 2
pv: /dev/loop5p4: No such file or directory
Writing partition 3
pv: /dev/loop5p3: No such file or directory
Writing partition 4
pv: /dev/loop5p4: No such file or directory
Writing partition 5
pv: /dev/loop5p3: No such file or directory
Writing partition 6
pv: /dev/loop5p6: No such file or directory
Writing partition 7
1.00GiB 0:00:45 [22.6MiB/s] [=======================================================================================================>] 100%            
Writing partition 8
pv: /dev/loop5p8: No such file or directory
Writing partition 9
pv: /dev/loop5p9: No such file or directory
Writing partition 10
pv: /dev/loop5p10: No such file or directory
Writing partition 11
pv: /dev/loop5p11: No such file or directory
Writing partition 12
32.0MiB 0:00:01 [22.7MiB/s] [=======================================================================================================>] 100%            
ChromeOS installed.

desmondam, 2020-06-18:
Thanks it works great.You are awesome

closed 2020-06-18 by desmondam