sebanc/brunch #321: Question: Updating the kernel

DiegoMagdaleno, 2020-06-16:
Is it possible to take the kernel 5.4 from Google for example, and add some brunch patches and use that?

I would be pretty interested into that, and i can look into providing a build for that.

sebanc, 2020-06-16:
Sure, you can build kernels from here:
https://chromium.googlesource.com/chromiumos/third_party/kernel/

The brunch patches are not even necessary (aside from some device specific features), you can already try to build one without the patches.

akhil-naidu, 2020-06-19:
Hey, @sebanc I would like to let you know one thing in which your involvement will make it a much better piece of work.

Just have a look at it, and I even explained what it does in this post itself (no need for extra research from you): https://github.com/dnschneid/crouton/wiki/Build-kernel-headers-and-install-Virtualbox-(x86)

In the above-attached link, the community tried to create something quite different from what the **Chromebook** is meant for. Iḿ emphasizing again that it is a Chromebook and not a ChromeOS. 

**Step 01**: They used Crouton to install a chroot based Linux system - pretty easy and the internet has a detailed guide about it.
**Step 02**: In the chroot Linux environment they configured the kernel so that they can install Virtual Box
**Step 03**: Now the possibilities that an individual can do increases a lot. Can install windows or Kali Linux for Pentesting and many.

**But the biggest drawback:** As it is a Chromebook, the amount of ram and available space is highly limited and the performance can be a disturbed one.

But now, you came into the game and cheated everything. You changed the basics of how ChromeOS can be used. 

> Frankly speaking, now we can install ChromeOS in a high-spec pc and the performance of this PC in gaming comparative to that of a Windows PC, particularly in CPU intense games -like CS: GO and DOTA- is unbelievable.

So, in this high spec PC installing a kernel that supports virtual box installation in a chroot instance will make it a perfect piece of machine for many others. I´m someone who is deep into cybersecurity and this kind of environment is what we need, highly flexible when in need (using chroot + VirtualBox) and highly rigid when you want to limit yourself to browsing + using play store in a PC.

**What is the problem, why I speak of all these?**
As mentioned earlier, the above procedure is meant for Chromebooks which are x86 and which have a ChromeOS based kernel.

**If you can provide me the right idea about how to proceed further** in order to change the kernel so that it can be compatible with VirtualBox, we can always create a brunch that can support virtualization in non-Chromebook based ChromeOS systems. I would also like to tell you that I will be available to you, always, if you want to focus on this part. If not, give me a lead.

sebanc, 2020-06-19:
I am typing this from my phone so it might not be the exact correct commands but the virtualbox kernel modules are dkms and if I remember well all you have to do to install virtualbox in crouton is to run:
`sudo mount -o remount,rw /lib/modules`
And then `sudo apt install virtualbox`

However, note that KVM can only be enabled in virtualbox if crostini is disabled in chromeos settings.

sebanc, 2020-06-19:
There was a typo in my previous post, I have just corrected it.

akhil-naidu, 2020-06-19:
The mount command gives me an error saying: "mount point not mounted bad option". For the command which was given after typo.

sebanc, 2020-06-19:
I have just ran it in crouton and it works for me, not sure what could be wrong here

akhil-naidu, 2020-06-19:
I have the installation of a virtual box, the problem is I cannot run a VM. Also, what image are you using (xenial or some other)?

akhil-naidu, 2020-06-19:
`sudo modprobe vboxdrv` this gives an error!, I mentioned it below

modprobe: FATAL: Module vboxdrv not found in directory /lib/modules/4.19.122-brunch-sebanc

sebanc, 2020-06-19:
I have it currently installed (and woring) with bionic + xfce, i had also tried xenial + xfce in the past and it was working.

1) remove virtualbox (it might be installed without modules being correctly built).
2) run the mount command and make sure it returns without error.
3) install virtualbox

akhil-naidu, 2020-06-19:
yeah sure, I will proceed as you mentioned. 

Btw, I found this syntax error. Have a look: https://askubuntu.com/questions/175739/how-do-i-remount-a-filesystem-as-read-write

sebanc, 2020-06-19:
I have to disagree, the command works and I have used it for years.

akhil-naidu, 2020-06-19:
Right now Iḿ using **kali-rolling** image. let me check it with your environment, iḿ going with `sudo crouton -r xenial -t xfce`

Most probably iḿ hoping this would solve my problem and I will update my results here as soon as I have some output or conclusion.

akhil-naidu, 2020-06-20:
It works, thanks to @sebanc. I concluded all this as a guide, In case if someone is still facing trouble in Installing VirtualBox in ChromeOS. [Click Here](https://github.com/sebanc/brunch/issues/330)

sebanc, 2020-06-20:
Thanks for the guide, it will probably help others, I just have 2 requests:
1) Please remove any reference to youtube videos. I am not related to those and they use other instructions/scripts which often create issues.
2) Could you add a note saying that crostini needs to be disabled in chromeos settings ? Else we will probably see people saying it is not working due to that.

akhil-naidu, 2020-06-20:
Sure, I will remove the links and scripts which are related to youtibe. 

Btw, I thought crosinit means Linux beta in chromeos. If I'm wrong let me know, as I mentioned in guide to disable Linux beta -as it worked for me.

sebanc, 2020-06-20:
Yes crostini means Linux beta, I must have missed it when I read the guide. Thanks for editing it :)

kowalski7cc, 2020-06-23:
I'm curios about @DiegoMagdaleno work if Kernel 5.4 could work better

DiegoMagdaleno, 2020-06-23:
@kowalski7cc I have some issues in boot last time I tried with Linux Kernel 5.4 it was stuck in "Disk size is now n", I wonder if @sebanc could give some insight

sebanc, 2020-06-26:
A kernel not booting in general means that the config is incorrect. Unfortunately, "Disk size is now n" would happen with any kernel and does not give indications on what might be incorrect in the config.

closed 2020-06-26 by sebanc