sebanc/brunch #322: Macbook Air 2012 iSight Camera Crashes

JPyke3, 2020-06-17:
Hi,

First of all I wanted to thank you for this awesome project. I really cant believe how well it works even on a MacBook. (Which always have tendancies to never play nice).

However, I have noticed that whenever I try to use the integrated iSight Camera on my 2012 MacBook air model. The camera seems to start for a split second. And then crash, after that the device will not be recognized. I am happy to provide any debug logs or output. I just didn't really know where to start with ChromeOS.

Cheers.

sebanc, 2020-06-20:
Hi,

Which recovery image do you use ?

JPyke3, 2020-06-22:
I used `rammus` as the recovery Image.

sebanc, 2020-06-22:
I think your laptop must have intel gen < 4, you should use "samus" it also has a better camera compatibility than "rammus" currently.

JPyke3, 2020-06-23:
Oh I do believe you are right. It's an Ivy Bridge processor which I completley overlooked. I will try with the `samus` image and report back.

JPyke3, 2020-07-06:
With that image camera works great 👍
Sorry for the delay in reporting back

closed 2020-07-06 by JPyke3