sebanc/brunch #324: Question: What brunch framework is installed

DennisLfromGA, 2020-06-17:
Just a question if I may.

I often, maybe too often ;-), install new brunch frameworks usually as soon as they're released but I can't seem to find a way to see the framework version that's installed.

I can see the release name with `uname -r`, (I.E.) 4.19.122-brunch-sebanc
But, the release .tar.gz files don't seem to show the release version number anywhere -
```
-rwxr-xr-x root/root     13172 2020-06-10 10:50 chromeos-install.sh
-rw-r--r-- root/root  33554432 2020-06-10 11:28 efi_legacy.img
-rw-r--r-- root/root  33554432 2020-06-10 11:28 efi_secure.img
-rw-r--r-- root/root 1073741824 2020-06-10 11:28 rootc.img
```
And looking at the files in partition 7 doesn't seem to give a clue to the framework version either unless of course, I'm missing it somehow -
```
-rw-r--r--. 1 root root   15 Jun 12 16:35 current_options
-rw-r--r--. 1 root root   11 Jun 12 16:35 debug_mode
-rw-r--r--. 1 root root 3.1M Jun 10 11:28 initramfs.img
-rw-r--r--. 1 root root 8.4M Jun 10 11:28 kernel
drwxr-xr-x. 6 root root 4.0K Jun 10 11:28 lib/
drwx------. 2 root root  16K Jun 10 11:28 lost+found/
-rw-r--r--. 1 root root   36 Jun 12 16:36 md5sum
drwxr-xr-x. 2 root root 4.0K Jun 10 11:28 packages/
drwxr-xr-x. 2 root root 4.0K Jun 10 11:28 patches/
drwxr-xr-x. 3 root root 4.0K Jun 10 11:28 usr/
```

I store the frameworks I've installed on the system -
```
ls -l /usr/local/brunch-work/
total 931M
-rw-r--r--. 1 chronos chronos 302M May 31 19:39 brunch_r83_k4.19_testing_20200528.tar.gz
-rw-r--r--. 1 chronos chronos 308M Jun  4 18:47 brunch_r83_k4.19_unstable_20200604.tar.gz
-rw-r--r--. 1 chronos chronos 321M Jun 12 16:02 brunch_r83_k4.19_unstable_20200612.tar.gz
```

So I usually just assume I have the latest one installed but if something goes awry with a framework version I'll usually install an earlier one. I sometimes forget which one is installed so being able to pull up the framework version would be helpful - to me at least.

I'd be grateful if you can point me to a way to find the framework version on the installed system but if that's not possible then don't worry about it, just asking.

Thanx,
-**DennisLfromGA**

dkay0670, 2020-06-18:
I have a question for my own, where are the unstable releases?

sebanc, 2020-06-20:
Hi, currently the best way to know which release you are using  is to look at the kernel build date in the first line of the "dmesg" output. However, I agree that it is not convenient and I am trying to think of something else to better keep track of the version in an automated way.

@dkay0670 For now latest testing release is the most recent but unstable releases are there:
https://github.com/sebanc/brunch-testing/releases

dkay0670, 2020-06-20:
> @dkay0670 For now latest testing release is the most recent but unstable releases are there:
> https://github.com/sebanc/brunch-testing/releases
@sebanc Oh, okay thanks!

dubagunta-rohit, 2020-07-03:
[image]

Go to Android preferences.. and then to about device... click on android version

DennisLfromGA, 2020-07-03:
@dubagunta-rohit 

Thanx, good to know. 👍

-**DennisLfromGA**

sebanc, 2020-07-04:
I have just uploaded a new release, brunch version is now available by running "cat /etc/brunch_version" in crosh shell.

DennisLfromGA, 2020-07-04:
@sebanc Thanx so much. 👍

closed 2020-07-04 by DennisLfromGA