sebanc/brunch #325: Thinkpad 8 Bailtray Atom No Wifi (bcm4324)

jwagnervaz, 2020-06-18:
Already tried the framework 》 options=broadcom
The wifi works at cloudready.. 

I have to say u do an amazing work.

Working:
GPU
Touchscreen
USB
Accelerometer
Android Apps
ACPI baterry
Mobile 3G module

Not working 
Wifi / Bluetooth  = combo bcm4324 
Sound

Thank you a lot.
You ar an genius.

Skawtnyc, 2020-06-18:
Are you sure you have that right? The correct setting is "options=broadcom_wl". Not what you typed.

jwagnervaz, 2020-06-20:
> Are you sure you have that right? The correct setting is "options=broadcom_wl". Not what you typed.

Yes I used the correct option

sebanc, 2020-06-20:
Hi,

Could you install the latest r83 release if you are not using it, run brunch without "options=broadcom_wl" and attach a dmesg to this issue (run "dmesg > ~/Downloads/dmesg.txt" and post dmesg.txt from your Downloads folder) ?

jwagnervaz, 2020-06-21:
> Hi,
> 
> Could you install the latest r83 release if you are not using it, run brunch without "options=broadcom_wl" and attach a dmesg to this issue (run "dmesg > ~/Downloads/dmesg.txt" and post dmesg.txt from your Downloads folder) ?

Thanks for your help. 
heres the log[
[dmesg.txt](https://github.com/sebanc/brunch/files/4808912/dmesg.txt)
](url)

sebanc, 2020-06-21:
Download the below file:
[brcmfmac43241b5-sdio.txt](https://github.com/sebanc/brunch/files/4808953/brcmfmac43241b5-sdio.txt)

Run `sudo cp brcmfmac43241b5-sdio.txt /lib/firmware/brcm/` in crosh shell.

Then reboot and check if wifi is working.

jwagnervaz, 2020-06-21:
> Download the below file:
> [brcmfmac43241b5-sdio.txt](https://github.com/sebanc/brunch/files/4808953/brcmfmac43241b5-sdio.txt)
> 
> Run `sudo cp ~/Downloads/brcmfmac43241b5-sdio.txt /lib/firmware/brcm/` in crosh shell.
> 
> Then reboot and check if wifi is working.

Worked Instantly, i cant say how amazing skills you have, an wizard genius!!!

Maybe i need a firmware to Bluetooth too? 
It appears to turn on as option but don't find anything.

My cameras and sensors are from Intel  i think, compatible with Linux, do i need some drivers ?
the cameras are 
Camera Sensor IMX175 and Sensor OV2722
Sensors didn't find the info, but i think its bay trail standard.

Thank you so so so so much.

sebanc, 2020-06-21:
> Worked Instantly, i cant say how amazing skills you have, an wizard genius!!!

Nice that it works, :) I will include this firmware directly in future releases.

> Maybe i need a firmware to Bluetooth too?
> It appears to turn on as option but don't find anything.

If you post a new dmesg, I will check if a firmware is missing but you are probably affected by the quite common "bluetooth working from command line but not from the settings menu" issue. (#331). ChromeOS r84 is supposed to bring new bluetooth improvements, I hope it will help with this. 

> My cameras and sensors are from Intel i think, compatible with Linux, do i need some drivers ?
> the cameras are
> Camera Sensor IMX175 and Sensor OV2722
> Sensors didn't find the info, but i think its bay trail standard.

Sorry but chromeos only supports usb cameras, even though your cameras might be supported in linux they will not work in chromeos. You can still try with "reef" or "samus" recovery images which are the most compatible for camera but I highly doubt it will change something.

jwagnervaz, 2020-06-21:
[dmesg.txt](https://github.com/sebanc/brunch/files/4809612/dmesg.txt)

heres the new log

sebanc, 2020-06-21:
Indeed, you miss a bluetooth firmware but to know which one I would need you to post the output of "lsusb"

jwagnervaz, 2020-06-21:
@sebanc  here it comes

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 0bdb:193e Ericsson Business Mobile Networks BV  (my 3g module) 
Bus 001 Device 004: ID 045e:0745 Microsoft Corp. Nano Transceiver v1.0 for Bluetooth (my usb keyboard and mouse)
Bus 001 Device 003: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

sebanc, 2020-06-21:
Could you unzip the below file in your Downloads folder, then run "sudo cp ~/Downloads/BCM.hcd /lib/firmware/brcm/", reboot and see if bluetooth works ?
[BCM.zip](https://github.com/sebanc/brunch/files/4809892/BCM.zip)

For battery drain, octopus is not the best recovery image due to a camera framework issue which increases cpu workload, try "reef" maybe.

jwagnervaz, 2020-06-21:
> Could you unzip the below file in your Downloads folder, then run "sudo cp ~/Downloads/BCM.hcd /lib/firmware/brcm/", reboot and see if bluetooth works ? [...]

Worked instantly again, you ar an magician!!!!
For me you can close the case.

I really appreciate an amazing human like you,  help me alot, thanks u so much, hope you have an amazing Sunday !

sebanc, 2020-06-21:
Unfortunately, I cannot add this firmware directly in brunch as the same firmware name is used for significantly different hardware. After each update, you will have to copy this firmware manually.

Have a nice sunday too !

closed 2020-06-21 by jwagnervaz

Z-SQL, 2021-06-24:
> Download the below file:
> [brcmfmac43241b5-sdio.txt](https://github.com/sebanc/brunch/files/4808953/brcmfmac43241b5-sdio.txt)
> 
> Run `sudo cp ~/Downloads/brcmfmac43241b5-sdio.txt /lib/firmware/brcm/` in crosh shell.
> 
> Then reboot and check if wifi is working.

Hi there, 

I have a similar issue and wifi and sound is not working 
wifi driver name: RTL8821 Ce 802.11 ac PCI wireless Network driver in my intel celeron n4020 cpu notebook

once I type the above commands to crosh shell: error:unknown command : sudo

 thanks for your amazing support