sebanc/brunch #329: ChromeOS is very laggy

desmondam, 2020-06-20:


dkay0670, 2020-06-23:
What GPU do you have?

desmondam, 2020-06-23:
Intel® HD Graphics for Intel Atom® Processor Z3700 Series

sebanc, 2020-06-23:
If I understand well, you were running chromeos from a usb flashdrive which is expected to be slow unless you have a usb 3.0 ssd flashdrive.

Nevertheless, adding "intel_idle.max_cstate=1" to the kernel command line seems to help improving performance on atom processors.

desmondam, 2020-06-26:
Ya after installing on hdd its better.still its not smooth sometmes

sebanc, 2020-06-26:
Ok I am not sure what your cpu is due to your 2 last comments, globally:
- if it is "Atom® Processor Z3700 Series", add "intel_idle.max_cstate=1 enforce_hyperthreading=1" to the kernel command line to have max performance.
- If it is "Celeron N2820", just add "enforce_hyperthreading=1" to have max performance.

If the performance is still not as good as you would expect, you can try other recovery images (notably reef or fizz) and see if it helps.

As far as I know, aside from that there is not much to be done to improve performance on those devices.

desmondam, 2020-06-26:
Thanks for your reply. But i don't know where to add this line. Can you please help me.

hbarsaiyan, 2020-06-26:
@desmondam refer to this https://youtu.be/db4Xo_xPTR4 and instead of 
```
enable_updates
```
add the required options

desmondam, 2020-06-27:
Thank you very much @hbarsaiyan

closed 2020-06-27 by desmondam