sebanc/brunch #330: GUIDE: How to Install Virtal Box in ChromeOS

akhil-naidu, 2020-06-20:
In this guide, I assume you are familiar with @sebanc  [brunch](https://github.com/sebanc/brunch), [Crouton](https://github.com/dnschneid/crouton), [Chroot](https://www.youtube.com/watch?v=2wSJREC7RV8) and obviously [Virtualbox](https://www.virtualbox.org/). If you are not familiar with any one of these first go through the following links and get a brief idea for conceptual clearance in your head. 

These are the steps we cover in this Guide.
- [ ] Install ChromeOS in your PC

- [ ] Installing a Chroot based Ubuntu Instance

- [ ] **Setting up the Environment in Ubuntu**

- [ ] Installing Virtual Box

### Installing ChromeOS in your PC
I guess you are familiar with the Brunch of this master branch, with which you can use Install the ChromeOS in any PC - provided your PC meets the [hardware requirements](https://github.com/sebanc/brunch#hardware-support-and-added-features). In case you have no idea how to follow the original documentation, I would suggest you follow these video tutorials - [Removes your current OS](https://www.youtube.com/watch?v=4gZYV0RWJQ8) and [For Triple boot](https://www.youtube.com/watch?v=FEY9MtznYGU&t).

### Installing a Chroot based Ubuntu Instance
Follow these instructions, in case of any errors stop it - do some research or comment it below.

1. Download crouton
2. Open a shell (Ctrl+Alt+T, type `shell`, and hit enter)
3. Copy the installer to an executable location by running `sudo install -Dt /usr/local/bin -m 755 ~/Downloads/crouton`
4. Now that it's executable, run the installer itself: `sudo crouton -r xenial -t xfce`
5. Wait patiently and answer the prompts like a good person. At the end of installation you need to provide username and password for ubuntu installation.
6. Done! You can jump straight to your Xfce session by running `sudo enter-chroot startxfce4` or, as a special shortcut, `sudo startxfce4`
7. Cycle through Chromium OS and your running graphical chroots using **Ctrl+Alt+Shift+Back(F1)** and **Ctrl+Alt+Shift+Forward(F2)**.
8. Exit the chroot by logging out of Xfce.

Step 6,7,8 are optional steps, which will provide you an idea about the usage of [crouton](https://github.com/dnschneid/crouton)

### Setting up the Environment in Ubuntu
After successful completion of Step 5, you can launch the Ubuntu using Xfce using Step 6 or you can use this command: `sudo enter-chroot` to access the Ubuntu terminal. It's up to you how you want to access the Ubuntu terminal.

In **ubuntu terminal**, use this command before installing anything. `sudo mount -o remount,rw /lib/modules/* `

### Installing Virtual Box
The above step, which focuses on setting up of environment is a must. The command: `sudo mount -o remount,rw /lib/modules/* ` should be executed without any error. **If you have no error proceed further**.

`sudo apt install virtualbox` to install VirtualBox
Before using VirtualBox, run `sudo modprobe vboxdrv`

We are all set, now you have a working VirtualBox in ChromeOS, If you have any suggestion or issue feel free to comment.

dkay0670, 2020-06-22:
VirtualBox In the title is misspelled.

FcoMarcosMAbreu, 2020-06-23:
Very good! It's possible to run VMware and snaps too?

WindowsXP95, 2020-06-24:
> In this guide, I assume you are familiar with @sebanc [brunch](https://github.com/sebanc/brunch), [Crouton](https://github.com/dnschneid/crouton), [Chroot](https://www.youtube.com/watch?v=2wSJREC7RV8) and obviously [Virtualbox](https://www.virtualbox.org/). If you are not familiar with any one of these first go through the following links and get a brief idea for conceptual clearance in your head.
> 
> These are the steps we cover in this Guide.
> 
> * [x]  Install ChromeOS in your PC
> * [x]  Installing a Chroot based Ubuntu Instance
> * [x]  **Setting up the Environment in Ubuntu**
> * [x]  Installing Virtual Box
> 
> ### Installing ChromeOS in your PC
> I guess you are familiar with the Brunch of this master branch, with which you can use Install the ChromeOS in any PC - provided your PC meets the [hardware requirements](https://github.com/sebanc/brunch#hardware-support-and-added-features).
> 
> > Any quires in this section, you are free to browse through the existing issues or to open a new issue.
> 
> ### Installing a Chroot based Ubuntu Instance
> Follow these instructions, in case of any errors stop it - do some research or comment it below.
> 
> 1. Download crouton
> 2. Open a shell (Ctrl+Alt+T, type `shell`, and hit enter)
> 3. Copy the installer to an executable location by running `sudo install -Dt /usr/local/bin -m 755 ~/Downloads/crouton`
> 4. Now that it's executable, run the installer itself: `sudo crouton -r xenial -t xfce`
> 5. Wait patiently and answer the prompts like a good person. At the end of installation you need to provide username and password for ubuntu installation.
> 6. Done! You can jump straight to your Xfce session by running `sudo enter-chroot startxfce4` or, as a special shortcut, `sudo startxfce4`
> 7. Cycle through Chromium OS and your running graphical chroots using **Ctrl+Alt+Shift+Back(F1)** and **Ctrl+Alt+Shift+Forward(F2)**.
> 8. Exit the chroot by logging out of Xfce.
> 
> Step 6,7,8 are optional steps, which will provide you an idea about the usage of [crouton](https://github.com/dnschneid/crouton)
> 
> ### Setting up the Environment in Ubuntu
> After successful completion of Step 5, you can launch the Ubuntu using Xfce using Step 6 or you can use this command: `sudo enter-chroot` to access the Ubuntu terminal. It's up to you how you want to access the Ubuntu terminal.
> 
> In **ubuntu terminal**, use this command before installing anything. `sudo mount -o remount,rw /lib/modules/* `
> 
> ### Installing Virtual Box
> The above step, which focuses on setting up of environment is a must. The command: `sudo mount -o remount,rw /lib/modules/* ` should be executed without any error. **If you have no error proceed further**.
> 
> > **crostini** needs to be disabled in ChromeOS settings
> 
> 1. `sudo apt install virtualbox` to install VirtualBox
> 2. Before using VirtualBox, run `sudo modprobe vboxdrv`
> 
> We are all set, now you have a working VirtualBox in ChromeOS, If you have any suggestion or issue feel free to comment.

Thanks this is nice, but dont think this is the place for it. Ill add it to the wiki and give you credit

akhil-naidu, 2020-06-26:
> VirtualBox In the title is misspelled.

Thank you for your decent intimation, I corrected it.

dkay0670, 2020-06-26:
Nice

akhil-naidu, 2020-06-26:
> Very good! It's possible to run VMware and snaps too?

Yes, it is, but there are some minor limitations, but they usually don't affect many users. I might provide a list of such limitations in future guides.

akhil-naidu, 2020-06-26:
@WindowsXP95 

I really appreciate it. Also, if possible, attach the link of the wiki here so that I can use it as a reference if required.

JakunGamer, 2020-07-24:
After a reboot
I can't start any virtural machines anymore
I try to run 'sudo modprobe vboxdriver' but it said: modprobe: FATAL: Module vboxdriver not found in directory /lib/modules/4.19.122-brunch-sebanc
I think because I forgot to mount /lib/module, so I run: 'sudo mount -o remount,rw /lib/modules/*' but it said: 'mount: /usr/lib/modules/5.4.0-42-generic: mount point not mounted or bad option.'
Can anyone help me?

akhil-naidu, 2020-07-26:
You need to uninstall and install vbox again, without missing any of the above commands.

For now this is the only solution I can suggest.