sebanc/brunch #331: Main issues on which help is needed

sebanc, 2020-06-20:
Currently, the most common issues with brunch for which could probably be fixed are:
- Internal microphone not working out of the box but that can be enabled by using alsa commands (#56)
- Bluetooth working from command line but not from the settings menu.
- Some USB cameras not working with "rammus" recovery image (but working correctly with "reef" or "samus" recovery images).

Unfortunately, I cannot reproduce those issues on my hardware and I feel like I am going nowhere.

Any pointer/help on those would be greatly appreciated.

akhil-naidu, 2020-06-20:
Yeah, waiting for something like this. 

I would like to know something specific here, which I never tried. Right now I'm using **rammus** recovery and I want to know whether I can change my recovery image to **samus** using `sudo chromeos-update -r < path to the ChromeOS recovery **sammus** image > -f < path to the Brunch release archive >` without any errors?

sebanc, 2020-06-20:
@akhil-naidu In general it is possible, however some cases have been reported where login with an existing user account was not possible after switching recovery image.

jhgxgAHS, 2020-08-12:
microphone not working on dell inspiron 5458

sar777-dev, 2020-11-04:
But in my hp x360 14 laptop ;Internal microphone working out of the box' from brunch r83, which was first i used.

asukumar10248, 2021-04-20:
after running storage_test_2 and memory_test my laptop is now running fast and apps are also loading very fastly but is works in hdd only not in pendrive or usbstick

asukumar10248, 2021-04-20:
To use this method 
1 ctrl+alt+t 
2 type memory_test
3 type storage_test_2
but use it after startup without opening any thing and be patient this process will take approx 5 hrs then restart your laptop

asukumar10248, 2021-04-20:
> Currently, the most common issues with brunch which could probably be fixed are:
> 
> * android apps not starting unless the user wait 1/2 minutes on the login screen.
> * Internal microphone not working out of the box but that can be enabled by using alsa commands (#56)
> * Bluetooth working from command line but not from the settings menu.
> * Some USB cameras not working with "rammus" recovery image (but working correctly with "reef" or "samus" recovery images).
> * Porting other sensors than Intel ISH to chromeos (bmc150, kxcjk-1013...)
> 
> I cannot reproduce those issues on my hardware and I feel like I am going nowhere.
> 
> Any pointer/help on those would be greatly appreciated.
After running storage_test_2 and memory_test my laptop is now running fast and apps are also loading very fastly but is works in hdd only not in pendrive or usbstick 
To use this method 
1 ctrl+alt+t 
2 type memory_test
3 type storage_test_2
but use it after startup without opening any thing and be patient this process will take approx 5 hrs then restart your laptop

cetheslick, 2021-04-22:
This may be noted elsewhere, but there is some movement on drivers for Surface cameras.
https://github.com/linux-surface/linux-surface/wiki/Camera-Support
https://github.com/linux-surface/linux-surface/issues/91

MadhiasM, 2021-05-19:
Hi @sebanc,

I have the 2-in-1 convertible Teclast F6 Pro with the Kionix KXCJ Sensor. Tablet mode detection and screen orientation does not work as with the other people in #89 or #808. However, I am able to retreive the accelerometer sensor data in the shell using one of the following commands:

Device 0 (Keyboard/Base accelerometer?)
cat /sys/bus/iio/devices/iio\:device0/in_accel_x_raw 
cat /sys/bus/iio/devices/iio\:device0/in_accel_y_raw 
cat /sys/bus/iio/devices/iio\:device0/in_accel_z_raw 

Device 1 (Screen accelerometer?)
cat /sys/bus/iio/devices/iio\:device1/in_accel_x_raw 
cat /sys/bus/iio/devices/iio\:device1/in_accel_y_raw 
cat /sys/bus/iio/devices/iio\:device1/in_accel_z_raw 

When I tilt my laptop base or screen in certain direction, I can manipulate the values accordingly.

I don't know if this is any help to getting the sensors to work with Chrome OS but at least it seems that they are actually delivering raw data. I have attached all the files that are within the folders and some examples of the values. If you know how to further investigate please let me know

[Kionix_Log.txt](https://github.com/sebanc/brunch/files/6511061/Kionix_Log.txt)

ksoham622, 2022-02-03:
@sebanc 
if you enable fast connect and bluetooth revamp from chrome flags bluetooth seems to work fine.

sebanc, 2022-02-04:
> @sebanc if you enable fast connect and bluetooth revamp from chrome flags bluetooth seems to work fine.

Thanks for the tip ! it is very interesting. I cannot really enable it by default cause it might break things for some without bluetooth issues but it most likely means Google will push this change in the next releases.

ksoham622, 2022-02-06:
https://github.com/sebanc/brunch/pull/865#issue-805458037
@sebanc check I think you already know about this but is it possible for you to implement that ?
Just saw some issue about android apps not working.

JaniszewskiM, 2022-09-11:
I found a solution that works for me (Thinkpad T460)
Android Apps not working in dark mode - disabling it solves the problem.

Rstment25, 2023-02-18:
@sebanc Can I assist you somehow in figuring out why apps don't work without waiting on login screen ? ( I can't start play store or games most of the time - I login without waiting enough and then gotta reboot and wait again.... )

Like remote access to PC or something? Let me know  ...