sebanc/brunch #334: WIFI is not working in Dell Inspiron 3543 Intel 5th gen.. Broadcomm- (BCM943142HM)

subash-uduwaka, 2020-06-22:
There is wifi issue in my device. First time boot the wifi button was hidden. After i entered "options=broadcom_wl" then button appeared but connections are not showing. 
[image]

sebanc, 2020-06-22:
Hi,

Could you attach a dmesg to this issue with "options=broadcom_wl" enabled ? (run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and attach dmesg.txt from your Downloads folder)

Also, could you post the output of "sudo rfkill list all" ?

Lastly, do you have secure boot enabled ?

subash-uduwaka, 2020-06-23:
Yes i added "options=broadcom_wl" option but it didn't worked.
I have turned secure boot also
ok i'll post the output
"But i dont know how to attach dmesg.txt"

sebanc, 2020-06-23:
> "But i dont know how to attach dmesg.txt"

Drag and drop it from the file manager to the github comment area

dandax, 2021-01-09:
I’m having a similar issue. Added `options=broadcom_wl`, wifi is now enabled but can’t see any networks.

I’m also not able to open the crosh shell. Didn’t change anything in the boot other than adding `options`. Do I have to add an account first? Can’t even get passed initial setup without wifi.