sebanc/brunch #337: Crostini Inconsistencies

dkay0670, 2020-06-22:
Crostini Is working very inconsistently when enabling, requiring to press Retry a ton to try to get it to enable. It downloads and installs fine, but get error 13 when attempting to start the container.
[image]

dkay0670, 2020-06-22:
I will also note that I have tried looking up error 13, but nothing that was actually error 13 came up...

sebanc, 2020-06-23:
Hi,

Could you try to install crostini, wait for it to fail, then run "cp /var/log/messages ~/Downloads/msg.txt" in crosh shell and attach msg.txt from your downloads folder to this issue ?

dkay0670, 2020-06-23:
[msg.txt](https://github.com/sebanc/brunch/files/4821822/msg.txt)

ralphwig, 2020-06-24:
I had a similar issue and I don't think it told me but I didn't have enough space. Did you do the "sudo resize-data" command in the shell?

sebanc, 2020-06-24:
Is it a log from a crostini install or when trying to launch crostini ?

dkay0670, 2020-06-24:
It is the end of /var/log/messages

dkay0670, 2020-06-24:
I think it's when it's installing

sebanc, 2020-06-25:
This issue is very weird and I don't understand why you are the only one having it. As far as I can tell it seems to be related to the network/internet/router. Could you try to use install crostini while connected to your phone in tethering mode instead of your normal wifi/lan ?

dkay0670, 2020-06-25:
Yeah, that worked... Now the question is why doesn't it work on my network?

sebanc, 2020-06-25:
I am not very good at debugging networking issues, I would say it is something related to the DHCP server not working properly but that's just a guess. Or maybe you have mac adress / IP based filtering setup ?

dkay0670, 2020-06-25:
Okay, thanks for the help!

closed 2020-06-25 by dkay0670