sebanc/brunch #338: Thinkpad X250 Sleep Not Working Correctly

herineutrino, 2020-06-22:
Hi, First off, thanks for this project!. I've just setup on my Thinkpad X250 however sleep isn't working correctly. Once the device goes to sleep I'm unable to resume blank screen and no activity. The only way to recover is by holding down the power button and restarting. I've tried enabling the S3 suspend option but the result is still the same. Tried with different chrome os image (rammus r80, r81, r83), (eve r80, r81, r83) using the latest brunch release and the problem still exist.

sebanc, 2020-06-23:
Hi,

Could you try disabling TPM in the bios if you have that option ?

herineutrino, 2020-06-23:
Thank you very much, after disabling security chip in bios sleep is working properly.

closed 2020-06-23 by herineutrino