sebanc/brunch #344: Netflix still not working after removing cros_debug in kernel line?

b1t0y, 2020-06-24:
I can now see netflix and disney app on play store after removing cros_debug and powerwash but still getting there is a problem playing. Please try again (5.8) error.  Is there any fix to it? thanks

sebanc, 2020-06-29:
Hi,

Could you try the latest release here ? (validate the github email invite first)
https://github.com/sebanc/brunch-testing/releases

b1t0y, 2020-06-29:
Thanks.  I will try it later when I get home.  Thanks for looking at this issue.  Will report back.

b1t0y, 2020-06-29:
Hi I tried the new brunch and powerwash but till getting the same 5.8 in netflix. Playstore also was slow during no cros debug on kernel.  Do I need to install the stable brunch or can maintain with this testing brunch?  Thanks for trying to solve the problem.  I am using your brunch in all my computer with dual booting and works really great.

sebanc, 2020-06-30:
What is your computer model ?

b1t0y, 2020-06-30:
I have a lenovo u430 i5 4th gen with intel hd4400.  Have you tried this on your PC/laptop and it worked?  I have other laptops that I can test with such as an atom and 3rd gen intel 3000 and 4th gen intel hd4600 if you think there will be a difference with gpu or model.  Thanks again will keep using the brunch and update the others.

sebanc, 2020-06-30:
Hi, I can confirm it works on my Surface Go (tried in the past and also yesterday) but I only tested direct video playback and forgot to test on downloaded content.

b1t0y, 2020-06-30:
> Hi, I can confirm it works on my Surface Go (tried in the past and also yesterday) but I only tested direct video playback and forgot to test on downloaded content.

So what step did you do?  You are using the netflix app download on the playstore? What I did was to update the brunch with this one.  Then after updating power washed it then boot without cros debug on the kernel.  Then after, I can see netflix on playstore and downloaded it.  Is that right? Thanks

sebanc, 2020-07-01:
Your steps (removing cros_debug from kernel command line and powerwash) are correct. At first I thought it could be a selinux issue but apparently it it not. Which image are you using and are you on beta or dev channel ?

b1t0y, 2020-07-01:
> Your steps (removing cros_debug from kernel command line and powerwash) are correct. At first I thought it could be a selinux issue but apparently it it not. Which image are you using and are you on beta or dev channel ?

Im using rammus 83 with your testing brunch you gave me.  I did not change the channel when it was first installed.  So I guess that is the stable channel?  Do I need to change the channel?  Do you think I should do a clean install with the no cros_debug line when I boot the first time?

sebanc, 2020-07-01:
Ok, don't change channel, i was aking because netflix is often broken on beta/dev channel.

I have disabled selinux which is the only thing I can think of that could prevent netflix from playing.

I found this by searching:
https://forum.xda-developers.com/honor-8x/help/netflix-error-5-8-t3914141

It seems this error can be related to other android apps, could you try removing other android apps or even extensions and see if it works ?

NSbuilder, 2020-07-06:
I recently installed Chrome OS using latest brunch. Can't find netflix on store... even after disabling cros_debug

sebanc, 2020-07-07:
@NSbuilder You also have to powerwash after removing cros_debug

NSbuilder, 2020-07-07:
Do I have to power wash before removing cros_debug? How do I powerwash?

NSbuilder, 2020-07-07:
> @NSbuilder You also have to powerwash after removing cros_debug. Also, just in case you would be on beta/dev update channel, safety net is almost always broken there.

Now play store is very laggy :(

NSbuilder, 2020-07-07:
> @NSbuilder You also have to powerwash after removing cros_debug. Also, just in case you would be on beta/dev update channel, safety net is almost always broken there.

I did. Now android apps are laggy to the point of unusable. What can I do?

NSbuilder, 2020-07-07:
b1t0y also said that play store was slow. For me it wasn't slow, it was freezing, app crashes... etc

I will try to reinstall brunch and removing cros_debug on first boot?

BuzzKnightbeer, 2020-07-15:
Just wanted to chime in here. I'm on stable, removed cros_debug and did a Powerwash. Can't see Netflix, Stadia, Geforce NOW, etc in the Play Store.
I'm on a Surface Pro 6 with rammus 83 and r83-k4.19-stable-20200708.

sebanc, 2020-07-15:
@BuzzKnightbeer You should at least see Netflix, don't know for the others. Which recovery image do you use ? When you press CTRL+ALT+T and type "shell", do you have an error ?

BuzzKnightbeer, 2020-07-15:
@sebanc I used chromeos_13020.87.0_rammus_recovery_stable-channel_mp-v2.bin.zip
There are no errors when I launch the shell.

sebanc, 2020-07-15:
Then, you did not properly remove cros_debug, if you use dual boot make sure you are modifying your Linux grub config.

NSbuilder, 2020-07-15:
> @BuzzKnightbeer You should at least see Netflix, don't know for the others. Which recovery image do you use ? When you press CTRL+ALT+T and type "shell", do you have an error ?

I see netflix but android apps and play store glitch and lag. it "moves" to next menu after hovering on window buttons ( minimaze, close etc )

b1t0y, 2020-07-15:
> > @BuzzKnightbeer You should at least see Netflix, don't know for the others. Which recovery image do you use ? When you press CTRL+ALT+T and type "shell", do you have an error ?
> 
> I see netflix but android apps and play store glitch and lag. it "moves" to next menu ( without animation ) after hovering on window buttons ( minimaze, close etc )

I have the same issue when launching play store that it hangs.  Turning on night light option will help to run it much better.  I don't know why but it helps with mine.  Do you get netflix to work and play videos without cros_debug?  I still get an error on mine saying video 5.8 error even after power washing.

Nytuo, 2020-07-15:
I have the Same issue as @NSbuilder 
for me, Netflix videos work fine @b1t0y .
Before removing the cros_debug all work fine.
I also experience internet connection errors.

BuzzKnightbeer, 2020-07-15:
> Then, you did not properly remove cros_debug, if you use dual boot make sure you are modifying your Linux grub config.

I'm running a dual boot setup with Windows 10. I don't know how to make it work so that I don't need the USB to boot but for now I removed the cros_debug in Grub of the USB stick. Is that not the right place?

sebanc, 2020-07-15:
@BuzzKnightbeer In theory yes, but if you were in non-dev mode you would not be able to enter crosh shell

BuzzKnightbeer, 2020-07-17:
@sebanc Ah, I understand. I was able to create the Grub setup in Windows with Win2Grub and checked that the cros_debug is not in there. Netflix is now showing up.
Stadia and GeforceNow unfortunately not. They are both Game Streaming services. Stadia is from Google, so it's weird that it's not showing...

ssj-gz, 2021-12-21:
@b1t0y 

Hi - did you ever have any luck with this? I'm having exactly the same issue. 

* Developer mode is off; did a power-wash after:
```
Google Chrome: Version 96.0.4664.111 (Official Build)  (64-bit)
Platform: 14268.67.0 (Official Build) stable-channel rammus
Channel: stable-channel
Firmware Version:
ARC Enabled: true
ARC: 7982014
Enterprise Enrolled: false
Developer Mode: false
```
* Trying to run `shell` from cros gives command not found, as expected &#10004;
* Oddly, all of **Netflix**, **Disney+** and **Amazon Prime Video** apps installed from the Play Store and ran without complaints from the very first boot i.e. with Developer Mode still enabled!
    * In fact, Disney+ and Amazon Prime Video let me download episodes and watch offline without a single complaint, with Developer Mode enabled!
* ... but Netflix app still gives the **5.8** error on any attempted playback, whether it's online or a downloaded video :(

I've seen some people claim that you need to sideload an old version of Netflix, but I'm running into #1303 (i.e. can't even start Linux) and the listed mitigation for that didn't work for me :(

Apart from that, I'm very impressed with Brunch so far - it's quite an achievement :)

closed 2026-04-21 by github-actions[bot]