sebanc/brunch #345: Tablet mode issues

KonstantinosDetsis, 2020-06-24:
Hello, I've been using brunch for the past month on my Surface Pro (1st gen) and I really like it although I have an issue. De-attaching the keyboard does not automatically put the device to tablet mode. As you requested here is [the output](https://pastebin.com/W58yX1ku) of "sudo udevadm monitor" and de-attaching and re-attaching the keyboard.
I am currently using the latest version of brunch (Brunch r83 k4.19 testing 20200614) and latest version of ChromeOS (83). Also here is my grub config and output of "uname -a" just in case you need it.

..cros_debug enforce_hyperthreading=1 options=enable_sensors,acpi_power_button

Linux localhost 4.19.122-brunch-sebanc SMP PREEMPT Sun Jun 14 11:23:46 CEST 2020 x86_64 Intel(R) Core(TM) i5-3317U CPU @ 1.70GHz GenuineIntel GNU/Linux

Should I also be using the samus recovery?

sebanc, 2020-06-25:
Hi,

Could you try tablet mode with the unstable release available below ?
https://github.com/sebanc/brunch-testing/releases

Yes for Intel Gen 3 samus is the right recovery.

KonstantinosDetsis, 2020-06-27:
Thanks! I installed it but it doesn't do anything. I installed it using chromeos-update. Do I need to install it from scratch? Also [here](https://github.com/sebanc/brunch/issues/304) you said that it takes long to boot. Do you mean the linux sort of boot or the chrome logo because my Surface takes a long time there?

sebanc, 2020-06-27:
chromeos-update should be ok, could you attach a dmesg to this issue ? (run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and attach dmesg.txt from you downloads folder).

The chromeos logo is displayed while kernel modules are being loaded, you can have the ui displayed earlier by using "options=no_ui_delay" (the modules will load in the background but it causes issues with android apps on some devices).

KonstantinosDetsis, 2020-06-27:
Here is the dmesg. Also I didn't have the issue with long ChromeOS logo boot before I updated to ChromeOS 83. @ #
[dmesg.txt](https://github.com/sebanc/brunch/files/4840881/dmesg.txt)

KonstantinosDetsis, 2020-06-27:
I think I may have installed the wrong brunch release because this one that I downloaded doesn't mention anything about tablet mode in the changelog. 
[image]

sebanc, 2020-06-27:
Hi,

You used the right release, it's just that I don't mention device specific patches in release notes.

I tried to fix the issue with the previous build and uploaded a new one. Could you try it ?

sebanc, 2020-06-27:
Also, r83 increased boot time (at chromeos logo) on purpose to improve android apps compatibility on some devices, it is normal that you need to use "options=no_ui_delay" to get the previous behavior.

KonstantinosDetsis, 2020-06-27:
I just installed it and it also doesn't seem to do anything. Do I need to enable some sort of option in the grub config? Here is a dmesg...
[dmesg.txt](https://github.com/sebanc/brunch/files/4841202/dmesg.txt)

sebanc, 2020-06-28:
I can see in the dmesg that the tablet mode module is correctly activated this time. Normally if you remove the type cover (not fold it), it should switch to tablet mode.

Also, I notice that you have a wireless mouse/keyboard connected which could prevent tablet mode from working:
```
[    2.519995] input: Compx 2.4G Receiver as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/0003:1D57:FA60.0001/input/input3
[    2.571663] hid-generic 0003:1D57:FA60.0001: input,hidraw0: USB HID v1.00 Keyboard [Compx 2.4G Receiver] on usb-0000:00:14.0-2/input0
[    2.573575] input: Compx 2.4G Receiver as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.1/0003:1D57:FA60.0002/input/input4
[    2.574230] hid-generic 0003:1D57:FA60.0002: input,hidraw1: USB HID v1.00 Mouse [Compx 2.4G Receiver] on usb-0000:00:14.0-2/input1
[    2.576040] input: Compx 2.4G Receiver System Control as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.2/0003:1D57:FA60.0003/input/input5
[    2.627610] input: Compx 2.4G Receiver Consumer Control as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.2/0003:1D57:FA60.0003/input/input6
[    2.628018] hid-generic 0003:1D57:FA60.0003: input,hidraw2: USB HID v1.00 Device [Compx 2.4G Receiver] on usb-0000:00:14.0-2/input2
```

If it still does not work after removing the wireless keyboard/mouse dongle, I will try another method.

KonstantinosDetsis, 2020-06-28:
Sadly after disconnecting the mouse dongle and rebooting just in case, it still does not work.

sebanc, 2020-06-28:
Ok, no worries, I will try something else

KonstantinosDetsis, 2020-06-28:
Ok, thanks!

sebanc, 2020-06-28:
Could you try the new build with "options=force_tablet_mode" added to the kernel command line (after cros_debug) ?

KonstantinosDetsis, 2020-06-29:
Finally, it works, thank you very much! It even works with the mouse dongle connected. Now I want to ask a few more questions. If I insert an SD Card to my Surface using the built-in slot, the SD card appears as a USB in ChromeOS and I had to rename my SD Card to "SD Card". Is there any way to fix this? Finally, is there any possible way I can make chrome://flags work on samus or change the flags any other way? Thanks!

sebanc, 2020-06-29:
Nice to see that it eventually works :)

For the sdcard, unfortunately I don't think I have a way to change it as it is chromeos interpretation of your device.

Regarding chrome://flags, there are several solutions:
- update to dev channel (enable updates and run `sudo update_engine_client --channel=dev-channel -update` in crosh shell)
- use chrome://flags from guest mode (it only works for global flags but not user specific ones)
- find the corresponding command line argument and add it manually to /etc/chrome_dev.conf

KonstantinosDetsis, 2020-06-29:
Ok, I will update to dev channel. But what will happen when brunch r84 comes out? Will options=force_tablet_mode still work for me?

sebanc, 2020-06-29:
Yes the Ssurface Pro tablet mode patch will be included in all future builds.

KonstantinosDetsis, 2020-06-29:
Ok, thanks for your help!

KonstantinosDetsis, 2020-06-30:
I decided to open this issue again because I discovered another problem with tablet mode. Auto-rotation doesn't work. "dmesg | grep ish-hid" doesn't report anything. I remember when I was using brunch r80 with options=enable_sensors and was in tablet mode it worked perfectly. After I updated to brunch r83 it stopped working. Any ideas why?

sebanc, 2020-07-01:
enable_sensors option has been removed, they are enabled by default. Could you try to remove no_ui_delay and see if rotation works ?

Also, could you post the output of "lsmod" ?

KonstantinosDetsis, 2020-07-01:
Removing no_ui_delay doesn't seem to work. I have also made sure I have auto-rotation turned on in ChromeOS.
Here is the output of lsmod: [lsmod.txt](https://github.com/sebanc/brunch/files/4856208/lsmod.txt)
What apps do you know that support auto-rotation in ChromeOS?

sebanc, 2020-07-01:
It's the interface that rotates in chromeos so from the moment you are in tablet mode, if you rotate the device, the interface should match the device orientation (though some apps/games are designed specifically for portrait / landscape and can block rotation)

The accelerometer driver seems correctly loaded, could you try removing options=force_tablet_mode and to reboot ?

Then go to tablet mode with CTRL+ALT+SHIFT+T and check if rotation is working ?

KonstantinosDetsis, 2020-07-01:
It still doesn't work after removing both no_ui_delay and force_tablet_mode from "options=". Something which I noticed which could be related is that when the device is in tablet the and type cover is removed the virtual keyboard still doesn't appear on the screen and has to manually be enabled from settings every time the device exits tablet mode. Also since I have AIDA64 installed here is what it shows. 
[image]

sebanc, 2020-07-02:
> It still doesn't work after removing both no_ui_delay and force_tablet_mode from "options=". Something which I noticed which could be related is that when the device is in tablet the and type cover [...]

This is most likely because of your wireless keyboard adapter being plugged in.

 Also since I have AIDA64 installed here is what it shows.
> [image]

The AIDA64 output is correct as for the loaded modules, I really don't understand why it does not work, could you verify that rotation is not locked in the settings area ?

KonstantinosDetsis, 2020-07-02:
After removing the mouse dongle the virtual keyboard appears fine so thanks for that. Auto-rotation is turned on.

sebanc, 2020-07-02:
I changed something in the new unstable build that could have an impact on this. Could you update to it ? (it also has the tablet mode patch included)
https://github.com/sebanc/brunch-testing/releases

KonstantinosDetsis, 2020-07-02:
I've updated to the latest brunch-testing release ~~and still doesn't work~~. Also with the latest brunch-testing boot speeds are much faster (it's basically as fast as no_ui_delay) so that's nice. Okay I just found something strange. After the device boots up with the keyboard connected if you enable tablet mode with CTRL + ALT + SHIFT + T and NOT remove the keyboard rotation works. If I remove the keyboard rotation just stops working. Attaching it again doesn't make rotation work either. A full reboot is needed.

sebanc, 2020-07-03:
Could you replace the content of /lib/udev/rules.d/99-tablet-mode.rules with the below lines, reboot and see if there is a difference ?
```
ACTION=="add", SUBSYSTEMS=="usb", ATTRS{idVendor}=="045e", ATTRS{idProduct}=="07a9", RUN+="/bin/bash -c 'echo 0 > /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode'"
ACTION=="remove", SUBSYSTEMS=="usb", ATTRS{idVendor}=="045e", ATTRS{idProduct}=="07a9", RUN+="/bin/bash -c 'echo 1 > /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode'"
```

KonstantinosDetsis, 2020-07-03:
The 99-tablet-mode.rules file doesn't exist so I created but it still didn't work. If I manually run "echo 1 > /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode" it says that the file doesn't exist.

sebanc, 2020-07-03:
Sorry forgot to mention that this method is to be used with "options=force_tablet_mode", I thought you still had it enabled

KonstantinosDetsis, 2020-07-04:
Well I added it and it just broke everything. On the login screen my wallpaper was flashing, the icons on the bottom (power off, guest mode, etc) were bouncing up and down. After entering my password it took like 2 minutes to login. After that I was clearly able to see that the device kept switching from desktop to tablet mode every half a second. De-attaching the keyboard stopped it. Had to force reboot (because everything was very laggy) and remove force_tablet_mode from grub config, boot the device and reboot again.

sebanc, 2020-07-04:
Ok, could you boot again with "options=force_tablet_mode", then run "sudo rm /lib/udev/rules.d/99-tablet-mode.rules" in crosh shell, reboot and confirm if rotation works or not ?

KonstantinosDetsis, 2020-07-04:
The only difference is that the device now boots in tablet mode and that when I remove the type cover it doesn't go to tablet mode. If I rotate the device while it is in tablet mode and I haven't removed the type cover since I booted the device, rotation works. If I remove the type cover, rotation just stops working.

sebanc, 2020-07-04:
It does not really make sense to me right now, the only thing that I can think of which could explain it would be that the accelerometer is actually in the type cover.

Could use a live usb linux distro and confirm if the rotation works both with type cover plugged and type cover removed ?

KonstantinosDetsis, 2020-07-04:
I tried in Linux Mint but it didn't work (probably because Linux Mint isn't meant to be used on tablet devices). I will now download and try to see if Android x86 will work with auto-rotation.

KonstantinosDetsis, 2020-07-04:
The exact same thing that is happening with ChromeOS is happening with Android x86. On Windows auto-rotation without the keyboard works fine.

sebanc, 2020-07-04:
Could you try adding "button.lid_init_state=open" on the kernel command line ?

KonstantinosDetsis, 2020-07-05:
It didn't work so I removed it. This is my current grub config: 
...cros_debug quiet enforce_hyperthreading=1 options=acpi_power_button,force_tablet_mode

sebanc, 2020-07-05:
It looks like it is a bios / kernel issue as android x86 suffers from the same bug but I cannot find any information on how to solve that online. What's weird is that I cannot even find any discussion on this issue. Was your SP1 updated to the latest bios ?

KonstantinosDetsis, 2020-07-05:
I really don't know. This is how my bios looks: 
[image]
(picture taken from youtube video)
At some point around the start of 2020, I remember Windows Update was installing Surface Pro Bios or something like that. As far as I know the Surface Pro 5 bios looks completely different. Something like this: 
[image]
Any ideas how to update my bios?

sebanc, 2020-07-05:
If you had an update in windows in 2020, you are on the latest bios. I am sorry but I am out of ideas on this one...

If you can reproduce this issue on a linux distro like ubuntu, you can try to see if someone on linux-surface (https://github.com/linux-surface/linux-surface) has any idea about it. (they don't support brunch or androidx86, so you need to be able to reproduce this issue on linux)

KonstantinosDetsis, 2020-07-05:
Ok I will see if it happens with ubuntu too. Now I will ask something unrelated to this issue. If I have Secure Boot disabled in my bios and boot the device, it does this before it goes to grub: 
[image]
If I enable secure boot and reboot the same thing happens. If I enable secure boot and select install default secure boot keys it boots like that but without the red which is what I want but then after a few seconds it shows this: 
[image]
There is no option to enroll key from disk.

sebanc, 2020-07-05:
I guess the red background when secure boot is disabled is normal. On my SP5 there is an unlock sign which is probably equivalent.

It is however very weird that with secure boot enabled it succeeds to boot grub without the brunch key being enrolled... You might want to enroll the key while you are in ubuntu by downloading it in the master branch of brunch repo and running "sudo mokutil --import brunch.der"

KonstantinosDetsis, 2020-07-05:
Any way to run mokutil from ChromeOS or crouton because I don't have an ubuntu usb available to me right now?

sebanc, 2020-07-05:
It should work from crouton i guess

KonstantinosDetsis, 2020-07-05:
Ok I did run mokutil --import brunch.der from crouton and it said "input password: ". I typed a password and then the device froze for around 10 seconds. I then rebooted and I was taken to I think it was called POK Manager. There was the option to enroll key from disk. I selected it hit EFI-SYSTEM and brunch.der. I then hit Enroll MOK and it asked for a password. I typed the password which I set earlier and rebooted. It worked but it still showed a red screen which is what I want to get rid of.

KonstantinosDetsis, 2020-07-05:
Here is what I was talking about.
[image]

sebanc, 2020-07-06:
That's exactly it but I have no idea why SP1 considers brunch certificate as invalid, it works perfectly on my SP5...

KonstantinosDetsis, 2020-07-09:
Ok I just tried live-booting Ubuntu on my Surface and auto-rotation works perfectly with and without the type-cover.

sebanc, 2020-07-11:
Ok so this issue is definetly within the kernel, however it may have its origin anywhere in the kernel ACPI, hid, sensors... so it would be very difficult to identify the correct fix for it.

Could you try the below build ? (after accepting the github invite)
https://github.com/sebanc/brunch-next/releases

KonstantinosDetsis, 2020-07-11:
After updating my device got stuck in a black screen with a mouse after the chrome logo. I used a usb stick and Developer Console(CTRL + ALT + F2) to downgrade to the latest brunch-stable build. Thankfully everything went well and the login screen appeared.