sebanc/brunch #346: Pixelbook (Eve) No sound or brightness control w/either Eve or Rammus base

ericksontx, 2020-06-24:
Have used Brunch wonderfully on other machines but there are two major showstopper issues with Google Pixelbook 2017 / Eve.

These are consistent on multiple Eve machines, using either Rammus or Eve base recovery images, v80-v83 images and matching 80-83 Brunch stable scripts.

1. no sound. Sound is absent, and volume slider has no effect.

2. screen brightness control is broken: It is either completely on or completely off, slider and hotkeys have no effect (keyboard mapping for the hotkey controls is also a mismatch, but that is a different, minor issue).
Essentially this screen brightness is 100% all the time, when trying to slide to lowest brightness, the screen turns off. Adjusting up from lowest brightness results in the screen being 100% brightness again. Battery life killer.

Thanks

WindowsXP95, 2020-06-24:
You may need to enable some framework options

ericksontx, 2020-06-24:
> You may need to enable some framework options

Thanks , I am unfamiliar so how would I go about enabling them, and finding the correct ones? Thanks

sebanc, 2020-06-25:
Well, let's try to fix brunch issues with pixelbook:
1) Try to download and copy the firmware available below in /lib/firmware.
https://github.com/EmbeddedAndroid/linux-eve/tree/master/lib/firmware
2) add "i915.enable_dpcd_backlight=1 module_blacklist=snd_hda_intel" to the kernel command line
3) reboot and see if there is any change for audio and brightness

ericksontx, 2020-06-27:
> Well, let's try to fix brunch issues with pixelbook:
>
> 1. Try to download and copy the firmware available below in /lib/firmware. [...]

Thanks for the tips, however I tried these, currently have that file in my /lib/firmware directory and also added that to the kernel line in grub, and no effect. Still no audio or brightness control unfortunately.

Since I am using Eve and not a generic x86 machine- what if I copied the files from /lib/firmware from the regular Eve install, to the Brunched version's /lib/firmware directory? Technically only the Eve-relevant firmware files need to be in there, I'm guessing?

sebanc, 2020-06-27:
Could you attach a dmesg to this issue ? (run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and attach dmesg.txt from you downloads folder)

ericksontx, 2020-06-27:
[dmesg.txt](https://github.com/sebanc/brunch/files/4840303/dmesg.txt)
[dmesg.txt](https://github.com/sebanc/brunch/files/4840304/dmesg.txt)

> Could you attach a dmesg to this issue ? (run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and attach dmesg.txt from you downloads folder)

sebanc, 2020-06-27:
The "i915.enable_dpcd_backlight=1 module_blacklist=snd_hda_intel" are not enabled in the dmesg you sent. Have you added those to your linux dual boot grub config ? Also, did you run "sudo update-grub" afterwards ?

ericksontx, 2020-06-28:
Ok sorry I had changed things but wasnt able to isolate instance/ use of grub with all of the dual config experimentation.

So now using cleaned HDD with no windows,and booting only off of Brunch USB, and making edits to that grub. No other chrome installs or other OS's on hard drive.

Attaching new dmesg. Not sure if I'm adding those modifiers to the right place in the kernel command line? But they are now appearing in that line.
Also added dfw_sst.bin to bin/firmware/

seems to be no changes in issues with backlight control and audio
[dmesg.txt](https://github.com/sebanc/brunch/files/4842959/dmesg.txt)

sebanc, 2020-06-29:
For sound, it looks better, could you try run the below command in crosh shell and reboot:
"sudo ln -s dsp_fw_kbl_v3420.bin /lib/firmware/intel/dsp_fw_kbl.bin"

If sound does not work, could you post a new dmesg ?

For backlight, it did not seem to help, I will try to see what the issue could be.

ericksontx, 2020-06-29:
> For sound, it looks better, could you try run the below command in crosh shell and reboot:
> "sudo ln -s dsp_fw_kbl_v3420.bin /lib/firmware/intel/dsp_fw_kbl.bin"
> 
> If sound does not work, could you post a new dmesg ?
> 
> For backlight, it did not seem to help, I will try to see what the issue could be.

Hi - I ran that in shell, and rebooted. On reboot, when I logged in, I went to check volume, the speaker logo was active and at least for the single volume key I was able to press, I saw the volume slider move - the speaker logo and the volume movement were both improvements over before, where there was neither.

However before I could check if there was actually sound output, there was an immediate reboot. This was only about 8sec after login. Subsequent reboots were the same, even if I did Guest. Whether I messed with volume or not, Chrome would reboot seconds later. I can no longer to crosh shell to save dmesg.

Is there a way to extract it anyway?

sebanc, 2020-06-29:
Press CTRL+ALT+F2 when you see the login screen, then login as "root" and type `rm /lib/firmware/intel/dsp_fw_kbl.bin`.

The firmware we used are probably not the right ones, I will try to get them from the eve recovery image directly.

ericksontx, 2020-06-29:
Ok update.
Started with a fresh Brunch drive - with the additions and edits you posted, no crashes or reboot this time, and consistent. 

Volume slider works as does mute button, in terms of on screen functionality, but still no sound at all.

Trying to attach but GitHub keeps giving a failure message and won't let me attach. Will post log  below.

[    0.000000] Linux version 4.19.113-brunch-sebanc (sebanc@tablet) (gcc version 8.3.0 (Ubuntu 8.3.0-6ubuntu1)) #1 SMP PREEMPT Sun May 10 13:29:05 CEST 2020
[    0.000000] Command line: BOOT_IMAGE=(hd0,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug i915.enable_dpcd_backlight=1 module_blacklist=snd_hda_intel
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000000fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000001000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007a6adfff] usable
[    0.000000] BIOS-e820: [mem 0x000000007a6ae000-0x000000007a72dfff] type 20
[    0.000000] BIOS-e820: [mem 0x000000007a72e000-0x000000007a7f1fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007a7f2000-0x000000007a7fcfff] usable
[    0.000000] BIOS-e820: [mem 0x000000007a7fd000-0x000000007a7fdfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007a7fe000-0x000000007aa83fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007aa84000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fd000000-0x00000000fe00ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed10000-0x00000000fed19fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fed84fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed90000-0x00000000fed91fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000047effffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.70 by EDK II
[    0.000000] efi:  ACPI 2.0=0x7aa96000  SMBIOS=0x7aa84000  MEMATTR=0x79493698 
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: Google Eve/Eve, BIOS MrChromebox-4.12 06/04/2020
[    0.000000] tsc: Initial usec timer 24106960
[    0.000000] tsc: Detected 1600.000 MHz processor
[    0.000012] tsc: Detected 1599.960 MHz TSC
[    0.000012] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000014] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000018] last_pfn = 0x47f000 max_arch_pfn = 0x400000000
[    0.000022] MTRR default type: write-back
[    0.000022] MTRR fixed ranges enabled:
[    0.000023]   00000-9FFFF write-back
[    0.000024]   A0000-BFFFF uncachable
[    0.000024]   C0000-FFFFF write-back
[    0.000025] MTRR variable ranges enabled:
[    0.000026]   0 base 007B800000 mask 7FFF800000 uncachable
[    0.000027]   1 base 007C000000 mask 7FFC000000 uncachable
[    0.000027]   2 base 0080000000 mask 7FC0000000 uncachable
[    0.000028]   3 base 00C0000000 mask 7FF0000000 write-combining
[    0.000029]   4 base 00D0000000 mask 7FF0000000 uncachable
[    0.000029]   5 base 00E0000000 mask 7FE0000000 uncachable
[    0.000030]   6 disabled
[    0.000030]   7 disabled
[    0.000030]   8 disabled
[    0.000031]   9 disabled
[    0.001055] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.001931] last_pfn = 0x7aa84 max_arch_pfn = 0x400000000
[    0.008289] Scanning 1 areas for low memory corruption
[    0.008294] Using GB pages for direct mapping
[    0.008297] BRK [0x460c01000, 0x460c01fff] PGTABLE
[    0.008298] BRK [0x460c02000, 0x460c02fff] PGTABLE
[    0.008299] BRK [0x460c03000, 0x460c03fff] PGTABLE
[    0.008317] BRK [0x460c04000, 0x460c04fff] PGTABLE
[... 830 lines cut ...]
[  115.075113] IPv6: ADDRCONF(NETDEV_UP): veth_peer_arc0: link is not ready
[  115.082197] arc_peer_arc0: port 1(veth_peer_arc0) entered blocking state
[  115.083772] arc_peer_arc0: port 1(veth_peer_arc0) entered disabled state
[  115.086167] device veth_peer_arc0 entered promiscuous mode
[  115.087801] arc_peer_arc0: port 1(veth_peer_arc0) entered blocking state
[  115.089338] arc_peer_arc0: port 1(veth_peer_arc0) entered forwarding state
[  115.097364] peer_arc0: renamed from peer_peer_arc0
[  115.105583] IPv6: ADDRCONF(NETDEV_CHANGE): veth_peer_arc0: link becomes ready
[  115.110857] arc_peer_arc0: port 1(veth_peer_arc0) entered disabled state
[  115.112707] device veth_peer_arc0 left promiscuous mode
[  115.114255] arc_peer_arc0: port 1(veth_peer_arc0) entered disabled state
[  118.659570] [3583] netd: blocked syscall 321
[  160.112952] Bluetooth: hci0: advertising data len corrected
[... cut ...]
[  161.244931] Bluetooth: hci0: advertising data len corrected
[  161.471958] Bluetooth: hci0: advertising data len corrected
[  161.691949] Bluetooth: hci0: advertising data len corrected
[  161.903925] Bluetooth: hci0: advertising data len corrected
[  162.118924] Bluetooth: hci0: advertising data len corrected
[  165.123035] bt_err_ratelimited: 13 callbacks suppressed
[  165.123038] Bluetooth: hci0: advertising data len corrected
[  165.341974] Bluetooth: hci0: advertising data len corrected

sebanc, 2020-06-29:
Could you try copying the content of the below archive to /lib/firmware ?
[eve_firmware.zip](https://github.com/sebanc/brunch/files/4847195/eve_firmware.zip)

sebanc, 2020-06-29:
and also copy the content of the below one to /usr/share/alsa/ucm/
[eve_ucm.zip](https://github.com/sebanc/brunch/files/4847237/eve_ucm.zip)

ericksontx, 2020-06-29:
> and also copy the content of the below one to /usr/share/alsa/ucm/
> [eve_ucm.zip](https://github.com/sebanc/brunch/files/4847237/eve_ucm.zip)

Ok I copied the files and directories of the two zips above, into the destination directories.

unfortunately audio went back to original state as regular Rammus base+Eve Brunch boot - speaker logo on volume control is inactive/"not complete", mute function works, volume keys have no effect on slider, and no sound output.

dmesg attached.
[dmesg.txt](https://github.com/sebanc/brunch/files/4847921/dmesg.txt)

sebanc, 2020-06-29:
Thanks, indeed a new firmware appeared in the log, could you copy it to /lib/firmware ?
[dsp_lib_dsm_core_spt_release.zip](https://github.com/sebanc/brunch/files/4848008/dsp_lib_dsm_core_spt_release.zip)

sebanc, 2020-06-30:
Alternatively, you can download the latest testing release below (after validating the github email invite):
https://github.com/sebanc/brunch-testing/releases

And boot with "options=eve" added to the kernel command line.

ericksontx, 2020-06-30:
> Thanks, indeed a new firmware appeared in the log, could you copy it to /lib/firmware ?
> [dsp_lib_dsm_core_spt_release.zip](https://github.com/sebanc/brunch/files/4848008/dsp_lib_dsm_core_spt_release.zip)

Unfortunately I get reboots after adding this, I cannot even get to login screen, I get kicked to UEFI shell. Is it supposed to be in lib/firmware or a subdir within perhaps?

sebanc, 2020-06-30:
Could you try the test build just to confirm if it is the same ?

ericksontx, 2020-06-30:
> Could you try the test build just to confirm if it is the same ?

Ok tried test build with Rammus 83 on Eve. Initial boot was normal and fine except for backlight control and audio of course. Added the Eve option to kernel command line then rebooted.

Seems same result, reboot shortly after Chrome logo splash screen. Login screen is there for a few seconds and then it reboots. Not enough time to get to Crosh to capture dmesg.

sebanc, 2020-07-01:
Ok so copying the original eve firmwares do not work... maybe there are specific kernel patches in eve kernel or it could be because eve kernel is version 4.4 if I am not mistaken.

I will try to find resources on how eve sound works in linux distros.

By any chance, is backlight control working in the new release ?

sebanc, 2020-07-01:
Just found out, there is no audio support for kabylake chromebooks in linux even from galliumos. So this one is going to be diificult....

ericksontx, 2020-07-02:
> Ok so copying the original eve firmwares do not work... maybe there are specific kernel patches in eve kernel or it could be because eve kernel is version 4.4 if I am not mistaken.
>
> I will try to [...]

No difference in backlight during the brief time I had to try backlight controls before it would reboot.

Also FYI - on the testing release there are definite improvements on DisplayLink compatibility.

Before, connecting a monitor to an attached DisplayLink adapter caused an automatic reboot. Now, works perfectly. Tried with two different model older HP DisplayLink adapters, both work great.

Crescendo-BLYAT, 2020-07-14:
offtopic: why would one use brunch on legit supported Chromebook?
Eve is still supported officially, right?

Resurrection2981, 2020-11-15:
> offtopic: why would one use brunch on legit supported Chromebook?
> Eve is still supported officially, right?

Dual boot Windows & Chrome OS.

ericksontx, 2021-05-29:
Closed.

Brightness control was fixed in regular brunch releases.

Sound still has issues but can be worked around by using legacy kernel as shown in this thread.

https://github.com/sebanc/brunch/issues/779

closed 2021-05-29 by ericksontx