sebanc/brunch #348: what is the best recovery image for Lenovo Ideapad u310/u410?

leosel, 2020-06-25:
what is the best recovery image for Lenovo Ideapad u310/u410?

WindowsXP95, 2020-06-25:
Which one do you own?

treypop, 2020-06-25:
https://github.com/sebanc/brunch#chromeos-recovery-images

> "samus" is the recommended image for devices with 3rd generation Intel CPU and older.

b1t0y, 2020-06-25:
I have the lenovo u410 touch and has been using samus recovery.  Everything is working well so far.  Added some frameworks like no_ui_delay, broadcom_wl, acpi_power_button and  advanced_als.  I added also enforce_hyperthreading=1.

sebanc, 2020-06-29:
Hi, I tried to clarify my recovery image recommendation in #359. I think the telegram group or discord server might be better to have other recovery images recommendations.

closed 2020-06-29 by sebanc