sebanc/brunch #350: Dual or triple boot using clover bootloader.

b1t0y, 2020-06-25:
Is there a way that I can boot the chrome os using the clover bootloader?  I can do it with grub but I have a mac os that only works with clover and not grub.  I have the chrome os on a separate HD but wanted to boot with just one boot loader.  Any help please and thanks in advance.

sebanc, 2020-06-26:
Hi, I think you can either chainload grub from clover or clover from grub.

To chainload clover from grub, you should just need to copy clover efi directory to the brunch efi partition (n°12) and use a config like:
```
menuentry 'macOS' {
chainloader /EFI/clover/CLOVERX64.efi
}
```

To chainload clover from grub, you will probably find more support on the telegram group or discord server.

b1t0y, 2020-06-30:
Thank you for this.  I was able to dual and triple boot with clover.

closed 2020-06-30 by b1t0y