sebanc/brunch #351: Fix for ALPS Touchpad not working

omminp, 2020-06-26:
Apologies if this has already been discussed and I missed it.

I have an older style laptop with ALPS touchpad and stick.  After boot with brunch, the stick works but the touchpad and its buttons do not work.  I followed  the advice in **#305** and the touchpad works in crouton but not in chromeos.  Changes to /etc/gesture/*.conf make no difference to touchpad behaviour.  So that was the end of the story.

Until I found this:  https://bbs.archlinux.org/viewtopic.php?id=248756

The "interim" fix here is to open a terminal in chromeos, and execute

" sudo echo 'options psmouse proto=imps' > /etc/modprobe.d/psmouse.conf "  without the quotes.  

Then either execute

" sudo modprobe -r psmouse ; modprobe psmouse "  and the touchpad will work, or alternatively reboot chromeos and the touchpad will work.

Note the touchpad works as a mouse with 2 buttons, and double tap to click works.  Gestures like scrolling do not work.  But it makes the laptop touchpad usable. I find that on my computer at least, the best mouse experience is obtained by going to chromeos settings, mouse, and making speed maximum and acceleration switched off.

People with Elantech or Synaptics touchpads may find the final suggested solution in the link works for them - unfortunately it doesn't work for me.

The psmouse.conf file is preserved on reboot, but it is lost when brunch recreates the system.  **@sebanc** maybe this fix could be added to the boot options.  I have no idea if there is a similar fix to the Elantech one that restores multitouch available for other touchpads, but inclusion of that (if available) would be great.

I'm happy to do any testing that may be of value.

sebanc, 2020-06-26:
Hi, thanks for working on it and sharing your progress, it is very helpful :)

My understanding on this would be that options psmouse proto=imps allows the touchpad to be correctly recognized by the cmt driver as gestures are working.

It is maybe now that the device is correctly recognized by cmt that modifying /etc/gesture/*.conf might be of interest to fine tune the configuration.

In the current state, I think adding "psmouse.proto=imps" to the kernel command line will be enough for the settings to be kept across updates. I will implement a specific patch including the psmouse module parameter part if at some point there is a specific cmt config to go with it.

Again, thanks for investigating and sharing your findings.

omminp, 2020-06-26:
I'll investigate /etc/gesture/*.conf to see if there is any improvement and post the results.

hbarsaiyan, 2020-06-26:
@omminp Hi my Dell Latitude E6320 also has an Alps Touchpad and stick. Adding these lines to /etc/gesture/40-touchpad-cmt.conf fixed it for me.
```
Option          "Touchpad Stack Version" "1"
Option          "Horizontal Resolution" "33"
Option          "Vertical Resolution" "33"
Option          "Integrated Touchpad" "1"
```
Maybe you can also try it and see if it fixes the Touchpad for you.

omminp, 2020-06-29:
**@hbarsaiyan**  Thanks for your comment - I have tried adding those options but I see no change in the touchpad behaviour. It would be appreciated if you could share the source of those options. I am trying to find a clue to what may work for me and without a starting point it is very challenging.

hbarsaiyan, 2020-06-29:
> **@hbarsaiyan** Thanks for your comment - I have tried adding those options but I see no change in the touchpad behaviour. It would be appreciated if you could share the source of those options. I am trying to find a clue to what may work for me and without a starting point it is very challenging.

https://github.com/hugegreenbug/xf86-input-cmt/blob/master/xorg-conf/50-touchpad-cmt-elan.conf this is where i got those lines from. You can try fiddling with the other configs to see if any works for you.
PS: You can check mouse behaviour after editing the file without doing a full reboot by doing chrome://restart

omminp, 2020-07-10:
I have done a lot of testing with variations in the .conf file both with and without  "psmouse proto=imps" , with no change in the trackpad behaviour.  What I did notice was that evtest with proto=imps finds 'PS/2 generic mouse' and lists events for the 2 buttons and cursor coordinates, but without proto=imps it finds 'AlpsPS/2 ALPS Dualpoint Touchpad' and many more events appropriate to a touchpad.  It seems the events are not interpreted properly, I see no cursor activity using a single finger, but using 2 finger scroll gesture when the cursor is on the screen, the cursor moves quickly off one side of the screen.
I am happy enough to have the trackpad working as a mouse using 'proto=imps'.  

Adding 'psmouse.proto=imps' to the kernel boot command line does makes the change survive brunch updates. Thanks for the support with this.

closed 2020-07-10 by omminp

jafer241177, 2020-08-09:
touchpad lenovo idealpad flex3