sebanc/brunch #353: Brunch USB fails to boot on Dell Optiplex 3060

Skawtnyc, 2020-06-27:
I've been enjoying Brunch quite a bit on my Surface Go. Today I did a test of Virtualbox under Crouton, and performance of running Windows in a VM under crouton was more than acceptable for my needs.

My Windows desktop system is a Dell Optiplex 3060 Micro with an 8th gen i5 processor, 8GB RAM, and 128GB SSD. More info here: https://www.dell.com/en-us/work/shop/desktop-and-all-in-one-pcs/optiplex-3060-micro-desktop/spd/optiplex-3060-micro

I made sure bitlocker was disabled and disabled TPM in BIOS. I have created the USB using both Rammus and Samus, with legacy EFI and without (using the -l switch). No matter what permutations I try, halfway through the boot sequence it fails with massive md5sum input/output errors (see attached image). I'm not sure where to go from here, and I can't even collect a dmesg txt file to diagnose what's happening. Any ideas?
[image]

sebanc, 2020-06-27:
Hi,

That's the first time I see this issue, I think the origin of the error happens prior to the md5sum. Could you make a short video of the boot process from grub to the moment md5sum fails ?

Skawtnyc, 2020-06-27:
I decided to give it one last attempt, going through the entire archive of issues first to see if there was anything that stood out. There were a number of tips for other issues that might have been similar, so here's the process I finally ended up with that DID WORK.

1) Disable TPM in BIOS
2) Wipe the partitions from the USB stick and create a single ext4 partition using fdisk on my Surface Go Brunch system.
3) Used the latest Brunch testing build, and the most recent (v83) build of Rammus,
4) Extracted Brunch using "tar zxvf" command.
5) Created chromeos.img file using standard command, did not use legacy efi switch.
6) Wrote chromeos.img to USB stick using Rufus 3.11.
6) Booted Dell Optiplex 3060 Micro system with new Brunch USB drive.

This time the linux boot took the normal long time it usually takes when first booting from USB. No errors, and I breathed a sigh of relief when the Chrome boot screen appeared. It works great. So now I've ordered faster/larger RAM (16GB, dual channel 3200) and a m.2 256GB NVMe drive with r/w speeds of 2100/1150MB/s. This should give me a system that can run Windows apps pretty fast under Virtualbox in crouton.

closed 2020-06-27 by Skawtnyc

sebanc, 2020-06-27:
Nice to see that you got it working :) Indeed with such a setup you should have very good performance in virtualbox

Skawtnyc, 2020-06-29:
Follow-up:

Apparently the problem came back when I installed the new m.2 drive, and that was after it worked just fine. It finally dawned on me that maybe the USB stick was bad, so I used a high speed MicroSD card in a USB adapter. That one worked fine and didn't exhibit those i/o errors. So I wiped the old USB stick and threw it away. It had a good run, but was no longer dependable.

The new setup is very fast, and will be even faster once I upgrade the RAM. The Microsoft ARC mouse also wakes up the system from sleep, which it didn't do on the Surface Go, and mouse scroll still works after sleep. I'm very pleased with how well this has turned out. I'll probably write an update after I've had a chance to set up Windows in Virtualbox and test some app performance.

Skawtnyc, 2020-07-02:
There are some compatibility issues with Steam and Battle.net. Some apps don't like the Virtualbox video drivers; some apps just crash the entire Windows VM using Ubuntu 16.04. I have workarounds so far involving an in-place upgrade of Ubuntu to 20.04, but that breaks sound and I'm working on fixing that.

The really amazing part is performance. The m.2 drive I got makes things ridiculously fast. For instance, the delay for Android apps being available due to the Android container still booting after login is now gone. By the time I've finished logging in the Android apps are already available. The init script I created to execute after Android finishes booting no longer needs ANY sleep timer. So nice.