sebanc/brunch #354: Samus image questions

clandestinz, 2020-06-27:
I have quite some questions about samus recovery image. Thanks for your time in advance

1. Why doesn't the flags page work on stable channel? I noticed that regardless of what chromeos version is running, `chrome://flags` will always show a blank page unless the device is running on dev channel.
2. Also, why can't we switch channels through chrome settings and have to resort to crosh shell?
3. June 2021 is the end of life for this recovery image. If this project will still be maintained by then, what other option will we have?

sebanc, 2020-06-28:
Hi,

Both 1 and 2 are related to the same TPM issue.

samus recovery image (and all old ones compatible with Intel Gen < 4 devices) use TPM 1.2 which does not work well with swtpm resulting in an ownership issue.
I tried different swtpm parameters combinations to make it work with TPM 1.2 but without success.

The dev channel is a specific case as it seems Google includes both TPM 1.2 and TPM 2 frameworks in dev channel builds which explains why there is no ownership issue.

As for point 3, this project uses chromeos recovery images and as such is only compatible with devices similar to the ones currently supported by Google. Seeing that Google still updates some devices that were supposedly AUE about 1 year ago, you can probably expect support on samus recovery image up until mid/end 2022 but that's just a guess.

In general, providing full chromeos for old devices unsupported by Google would be a totally different project similar to FydeOS where everything is built from source with added android apps layers (which is probably less legal so that's not really a direction I would be interested in going).

clandestinz, 2020-06-28:
So old recovery images have problems with swtpm. Apart from that, chrome os is a proprietary fork of chromium os and is not supported on non chromebook devices. Do you think this project will encounter legal issues?

sebanc, 2020-06-28:
I don't know but I try as much as possible not to redistribute proprietary work.

clandestinz, 2020-06-28:
Alright, thank you for your time

closed 2020-06-28 by clandestinz