sebanc/brunch #355: Trackpad does not work with Pixel 2013, everything else appears to be working good.  I had to use the sammus image for v83, couldn't get rammus to boot.

badrock21, 2020-06-27:


sebanc, 2020-06-29:
Hi,

Indeed as mentioned in the main readme, devices with Intel Gen < 4 have are not compatible with rammus, more on that here: #359 

Regarding touchpad, you will find the information you need in: #305

badrock21, 2020-06-29:
I got it working with Rammus and brunch v81. I am unsure as to why it works but I tried installing Windows 10 which works fine but no trackpad or touchscreen in there.  It tried coolstars drivers and some other i2c touchpad drivers to no avail.  Mr Chrombox seabios quit showing up during reboot.  I then went back to try b inch and now when I booted the USB drive the touchpad works full gestures and all.

badrock21, 2020-06-29:
Unrelated I have Rammus v83 working on an Acer c720 peppy as well which is also gen 3.

sebanc, 2020-06-30:
The Intel Gen < 4 rule does not apply to Celeron / Pentium processors (even though it is not properly documented as it could make things more confusing) so it is normal that Acer 720 works with rammus r83.

I am not sure what caused the touchpad to work, if it is only that it does not work with "samus" but works fine with "rammus" then you can try "lulu" recovery image instead which also supports Intel Gen < 4.

b1t0y, 2020-06-30:
> I got it working with Rammus and brunch v81. I am unsure as to why it works but I tried installing Windows 10 which works fine but no trackpad or touchscreen in there. It tried coolstars drivers and some other i2c touchpad drivers to no avail. Mr Chrombox seabios quit showing up during reboot. I then went back to try brunch and now when I booted the USB drive the touchpad works full gestures and all. It's installed to HD on the pixel 2013. Thanks for your work. If I can dump some data for you I'll be more than glad to do so.

So do you mean you were able to run rammus r83 on your pixel 2013 with brunch v81?  I have a Lenovo u410 with intel gen 3 like pixel 2013 specs.  I remember I was able to boot rammus at first,  tried to change to know what other images would work with my laptop.  When I was trying it to finally install rammus r83 in my laptop it would not boot and goes in like a bootloop.  May I know what combination you specifically used with your pixel 2013 to make rammus work?  Thanks so much.  Looking forward for your response.

badrock21, 2020-06-30:
Brunch v81 with with rammus v81 is working good on my pixel 2013.  The first time I installed rammus and brunch v83 it went into the boot loop and would never progress.   Sammus would boot but the track pad never worked.  

Google Chrome | 81.0.4044.141 (Official Build) (64-bit)
Revision | b3702a4a50f862440e43b48d17b4166b7941f211-refs/branch-heads/4044@{#1002}
Platform | 12871.102.0 (Official Build) stable-channel rammus

Firmware is Mr Chromebox latest June 2020.

uname -a
Linux localhost 4.19.113-brunch-sebanc #1 SMP PREEMPT Tue May 26 08:46:59 CEST 2020 x86_64 Intel(R) Core(TM) i5-3427U CPU @ 1.80GHz GenuineIntel GNU/Linux

Hope this helps you

closed 2020-12-03 by badrock21