sebanc/brunch #357: ChromeOS battery life discussion

sebanc, 2020-06-28:
Common issues which cause battery drain:
- Dual gpu are the most common cause of battery drain, the easiest way to fix it is to blacklist the dual gpu module (e.g. adding "module_blacklist=nouveau" to the kernel command line).
- IPU3 camera recovery images cause a battery drain due to the android camera service running in an endless loop in the background (ipu3 recovery images are notably nocturne, atlas, nautilus, soraka).
- The use of crostini causes battery drain: xalexis from telegram croissant group gets 1.5 hours more screen on time of a full charge with crostini disabled (but that may depend of the packages you have installed).

If you are not in the above cases, there are some basic things to consider when comparing battery life between ChromeOS and Windows/Linux:
- Do you have the same use in Windows and ChromeOS ? Clearly, you cannot compare the battery life with simple browsing in windows and playing android games in ChromeOS. Try to compare objective things like battery life when playing youtube videos in the browser or things like that.
- Do you have the same brightness setting in windows and chromeos ? Screen brightness is very high by default in ChromeOS which has a clear impact on battery drain.

Don't hesitate to discuss battery life here and to share your battery life related tips.

hbarsaiyan, 2020-06-29:
@sebanc I am using samus recovery and my laptop doesn't have a camera so will the camera framework keep running in the background? and if so is there any way of disabling it?

sebanc, 2020-06-29:
@hbarsaiyan "samus" recovery image is not ipu3, you should not have this issue regardless of whether your laptop has a camera or not.

midi1996, 2020-07-01:
HP Elite X2 G2 (detachable) experience:

- Battery life is quite good, around 5-6 hours of regular usage (like reading text online or browsing through images on the browser with a discord pwa -not android app- in the background running, around 30 tabs, some android apps are running in the background, brightness 50%, so around 250nits for my case, no crostini, I'm using crouton instead)
- Images I tried:
  - soraka (since it's also HP): 3-4 hours (battery is CONSUMED)
  - rammus: 5-6 hours (above)
  - nami: same as rammus (sometimes more sometimes less, depends on the usage)
- Settings:
  - Autobrightness is on, although after changing the brightness manually it wont re-activate until I put the tablet to sleep
  - BT and Wifi always on
  - Autorotation
  - Audio 50%
  - Brightness 50%
  - Using the testing branch of brunch
 
All in all, it is a very efficient OS, other linux distros like arch or ubuntu would run about 4 hours or less, macOS 4 hours if I'm lucky, it seems I have an issue with CPU power management in it, Windows can go up to 8 hours if I drop the display brightness to 10% (like using it in a dark room, only 5 tabs and probably onenote open, often it's 5-6 hours).

Lucaacer, 2020-07-05:
I waited for a week or so before posting, because I wanted to find the culprit for an abnormal battery drain that happens every or four days out of the blue when the device (which is a Lenovo Yoga 11e 5th Gen) is closed and should be in suspension mode... But I have no clue why such thing sometimes happens. 

A battery charge usually lasts for more than 10 hours, so it's very good. 

I just can't say why it sometimes drains 20-30% in 2 hours or so when the device, as I said above, is closed and should be quietly sleeping.

My guess is that thunderbird is still trying to sync, but I am not sure how I could find the evicence of such a behaviour.

hacker1024, 2020-07-13:
My battery life doubles from 4 to 8 hours on my Dell Inspiron 7586 (i7-8565U) on every image I've tried when I disable the Android components.

da0ist, 2020-09-02:
I've installed brunch on everything I own that's not a server and battery life is definitely an issue. The main machine I'm using daily right now is a T440p with Nvidia GPU. I just added module_blacklist=nouveau via /usr/sbin/edit-grub-config, hopefully I did it right.

I have an X220 also, but I'm not sure of the battery condition on that one. My work X1C6 doesn't seem to run as long on battery as with Ubuntu either as far as I can tell. I do the same thing on all of them, work (ssh, gcloud) and surf reddit. :-)

lupohirp, 2020-09-25:
Hi,

i share my kernel boot config set in grub: 
module_blacklist=nouveau modprobe.blacklist=nouveau nouveau.runpm=1 nouveau.noaccel=1 nouveau.nofbaccel=1 nouveau.modeset=0 i915.i915_enable_fbc=1 i915.i915_enable_rc6=1 i915.i915_lvds_downclock=1 pcie_aspm=force

with this i can get a lot of hrs with battery ( 6+ ) with android and linux container with an optimus laptop. 

i leave here so also @sebanc maybe can check if some are useless or maybe are useless

bisaalz, 2020-10-05:
> Hi,
>
> i share my kernel boot config set in grub:
> module_blacklist=nouveau modprobe.blacklist=nouveau nouveau.runpm=1 nouveau.noaccel=1 nouveau.nofbaccel=1 nouveau.modeset=0 i915.i915_enable_fbc=1 [...]

can u please tell me where to put the code to disable nvidia optimus in detail explanation. i am new

lupohirp, 2020-10-05:
> can u please tell me where to put the code to disable nvidia optimus in detail explanation. i am new

unfortunately it's not. Kernel doesn't have a module like bbswitch/bumblebee so you can't disable it. Also , if don't get wrong, it needs acpi_call and we don't have it. maybe sebanc will add in near future. with this, you can mitigate working on i915 gpu, adding some power tweaks.

sebanc, 2020-10-14:
I added the acpi_call module in brunch r86 which allows disabling the dgpu, you will need to:
- Identify the ACPI method which disables the dgpu on your device (in the DSDT)
- load the acpi_call module: "sudo insmod /lib/modules/acpi_call.ko"
- run the identified ACPI method:
echo '< ACPI method to disable the dgpu e.g. \_SB.PCI0.PEG1.GFX0.DOFF >' | sudo tee /proc/acpi/call 

Disabling the dual gpu can be automated at boot through the use of a bootscript (create an executable script in /var/brunch/bootscripts)

lupohirp, 2020-10-14:
> I added the acpi_call module in brunch r86 which allows disabling the dgpu, you will need to:
>
> * Identify the ACPI method which disables the dgpu on your device (in the DSDT)
> * load the acpi_call [...]

Hi @sebanc, how i can iddmesg|grep DSDT entify the ACPI method?
i did this in the shell :  dmesg | grep DSDT and shows this output : 

[    0.025963] ACPI: DSDT 0x000000008CFC4000 02089A (v02 XMCC   XMCC1703 00000000 ACPI 00040000)
[    0.791562] ACPI: \_SB_.PCI0.LPCB.EC0_: Used as boot DSDT EC to handle transactions
[    0.860427] ACPI: \_SB_.PCI0.LPCB.EC0_: Used as boot DSDT EC to handle transactions and events

should be this my method? _SB_.PCI0.LPCB.EC0_

bisaalz, 2020-10-18:
> I added the acpi_call module in brunch r86 which allows disabling the dgpu, you will need to:
>
> * Identify the ACPI method which disables the dgpu on your device (in the DSDT)
> * load the acpi_call [...]

Please Add \_SB.PCI0.RP01.PXSX Some Acer laptop Use this

sebanc, 2020-10-18:
@lupohirp The clearer guide I found on decompiling DSDT is:
https://mackonsti.wordpress.com/2012/02/13/extract-dsdt-using-ubuntu-live-cd/

Alternatively you can use the script which contains some common acpi methods to detect the one your device use:
https://github.com/mkottman/acpi_call/blob/master/examples/turn_off_gpu.sh

@bisaalz I am not adding any automated method to disable the dGPU because of this (found on bbswitch github and also the reason I will not include bbswitch):
```
On some machines, turning off the card is permanent and the card does not reappear on subsequents reboots, which can result into the screen staying black all the time, including the BIOS screen. If it occurs, first try to shutdown, unplug power cord, remove battery, wait 30s, then put everything back in and boot. If it's not solved, then the solution is to reset the BIOS to factory settings. Before executing bbswitch for the first time, it is therefore recommended to take note of the full key sequence in the BIOS to do a reset.
```

Once you have identified the method which works with your device and are confident that it does not cause any issue, create a script (ending by .sh and with execute permission) in /var/brunch/bootscripts directory (create it first if needed) with the below content:
```
insmod /lib/modules/acpi_call.ko
echo '< ACPI method to disable the dgpu e.g. _SB.PCI0.PEG1.GFX0.DOFF >' > /proc/acpi/call
```

sebanc, 2020-10-18:
Also, regarding the discovery of the correct ACPI method, someone with the same device might already have looked into it and posted the acpi method in the linux forums so a bit of googling might work too.

lupohirp, 2020-11-14:
for the battery life i've noticed a strange thing : if i don't enable crostini, i have a 5/6hrs barttery, that's quite reasonable. if i start crostini, my battery drop to 2 hrs, looks like that crostini enables discrete gpu ( i have an optimus laptop )

crodjer, 2021-01-15:
> My battery life doubles from 4 to 8 hours on my Dell Inspiron 7586 (i7-8565U) on every image I've tried when I disable the Android components.

How can I disable the android components @hacker1024?
I'd like to see if that improves my battery life, specially when suspended.

miaochenxi, 2021-08-09:
Will the gpu fan stop spinning instantly? Or will it just slow down?
My gpu is GTX 1650 and i uses \_SB.PCI0.PEG0.PEGP._OFF, the script works and the /run/proc/call shows 0x0called` . I think the function is successfully invoked but gpu fan still spinning crazily and my laptop's still hot for a long time.
It confuses me...

hacker1024, 2021-08-10:
> > My battery life doubles from 4 to 8 hours on my Dell Inspiron 7586 (i7-8565U) on every image I've tried when I disable the Android components.
> 
> How can I disable the android components @hacker1024?
> I'd like to see if that improves my battery life, specially when suspended.

@crodjer
I forgot how I did it, but I think it's just in the system settings somewhere.

crodjer, 2021-08-10:
@hacker1024 Thanks. Although, I switched to an official Chromebook now :-)

dyoghenes, 2021-09-28:
Hi!
I have a Lenovo notebook and the battery life is tragic... I only get 1-2 hour of battery life (using battery_test in 5 minutes it discharges 3.98%) 

I added these commands to the boot:
> i share my kernel boot config set in grub:
> module_blacklist=nouveau modprobe.blacklist=nouveau nouveau.runpm=1 nouveau.noaccel=1 nouveau.nofbaccel=1 nouveau.modeset=0 i915.i915_enable_fbc=1 i915.i915_enable_rc6=1 i915.i915_lvds_downclock=1 pcie_aspm=force

Does anybody have any idea what could be wrong? I am dualbooting with Windows 11, it can get up to 5 hours of battery life.

u9wy, 2021-10-02:
@sebanc 

Hi, I have a Dell XPS 13 2 in 1 2021 its an 11th gen Intel CPU and i'm running Volteer with the 5.10 Kernal. Experiencing pretty terrible battery life with Android and Linux both enabled and disabled. I seem to be getting random spikes from chrome sustaining high cpu usage from between 30% - 100% at times even after i've closed everything. However there do seem to be times when everything seems normal. During these times the discharge test is 0.20% in 30 seconds. 

Also sleep times are atrocious it seems as if the laptop is not sleeping after i close as there have have been several times where i've closed it and come back a few hours later and the laptop is hot with the fans running however there were other times when this didnt happen but every time the battery life on sleep has been terrible.

I'm also looking for a way to reduce the resolution from 3840 x 2400 to 1920 x 1200 to save more battery life. From the information I found I couldn't get anything working.

My grub config is below:

```
menuentry "ChromeOS" --class "brunch" {
        linux ($disk,7)/kernel-5.10 boot=local noresume noswap loglevel=7 cros_secure cros_debug video=1920x1080-32 intel_pstate=disable module_blacklist=nouveau options=enable_updates,internal_mic_fix,no_camera_config,pwa \
                console= vt.global_cursor_default=0 brunch_bootsplash=default quiet
        initrd ($disk,7)/lib/firmware/amd-ucode.img ($disk,7)/lib/firmware/intel-ucode.img ($disk,7)/initramfs.img
}
```

Thanks

biogon88, 2022-01-23:
Surface Go 2 with Pentium Gold 4425Y (possibly 8th gen?) running Rammus with 5.4 or 5.10 kernel. 

Standby seems to use roughly as much battery as having the screen on -- I get about 7-8% battery drain per hour of screen-off standby.

Tried using s3 kernel and it wouldn't resume from sleep. Any ideas? Can I help debug?

dustojnikhummer, 2022-01-29:
> Surface Go 2 with Pentium Gold 4425Y (possibly 8th gen?) running Rammus with 5.4 or 5.10 kernel.
>
> Standby seems to use roughly as much battery as having the screen on -- I get about 7-8% battery [...]

Surface Go 1 here, a very similar thing. I lose around 50% over night (so around your 7% per hour is quite accurate). In Windows 11 it seems to have disappeared after I changed the "networking connectivity in standby". It seems like, at least in Windows, during sleep the wifi card keeps pinging networks around it, causing this drain. Then it goes into hibernation mode. 

That doesn't seem to happen on ChromeOS. I can manually put it into hibernation (or some kind of deep sleep, it doesn't go through Grub again), which from my test of one night seems to work, but it is a keyboard shortcut. Which I can't do when I don't have a keyboard attach and just plop my Surface on my table when I stop using it before going to sleep.

srang992, 2022-06-25:
Today I installed the Chrome OS alongside windows 11. But it drains battery faster than windows 11. I have HP Pavilion x360 convertible laptop. I installed the volteer recovery image. Can anybody please help me in this matter?

tomsik-radek, 2022-06-25:
> Today I installed the Chrome OS alongside windows 11. But it drains battery faster than windows 11. I have HP Pavilion x360 convertible laptop. I installed the volteer recovery image. Can anybody please help me in this matter?

Yeah I noticed my Yoga C940 gets about an hour more in Windows 10. 
Are you talking about battery drain in idle/work or in sleep?

srang992, 2022-06-25:
> Yeah I noticed my Yoga C940 gets about an hour more in Windows 10. Are you talking about battery drain in idle/work or in sleep?

I notice this while in work. I also see how much it will drain battery in sleep mode, it seems that there is no problem while in sleep. How much battery life you get in your computer while using Chrome OS?

tomsik-radek, 2022-06-25:
> I notice this while in work. I also see how much it will drain battery in sleep mode, it seems that there is no problem while in sleep. How much battery life you get in your computer while using [...]

It's around 6.5 in Windows and 5.5 in ChromeOS. Though I do have forced hyperthreading On in Brunch, so it might be that?

ilykami, 2024-01-08:
ive been getting like 2h of battery life on an asus vivobook with 11th gen i5, any ideas why?

kurtbahartr, 2024-06-06:
Casper Excalibur G770.1245 gets slightly more battery performance on ChromeOS than Windows and regular Linux installations even with no specific tweaks regarding battery, which I would normally do on the other OSes.

I don't game much on Linux and I tend to game a lot on Windows but with a problem: When I game on either of the OSes, there would be significant performance drops from time to time when on battery since the dGPU, the Nvidia GeForce RTX 3050 Mobile GPU, would also be utilized. It's not the case with ChromeOS since the recovery I have, `volteer`/`voxel`, doesn't even have `nouveau` driver loaded OOTB and so the only GPU that's up and running is the iGPU, Intel UHD Graphics for 12th Gen Intel Processors.

I mostly use a mixture of Chrome and Android apps, the ones I always keep open are Spotify (Android, Play Store), WhatsApp (Android, Play Store), Discord (PWA, Stable), X (PWA) and Telegram WebA. If my laptop supported hybrid sleep and Brunch could also expose it to ChromeOS, I would also set some alarms on the Clock app (which is normally hidden away from the launcher but could easily be launched from the Play Store - [Play Store page for Google Clock](https://play.google.com/store/apps/details?id=com.google.android.deskclock) - or through Android settings (Apps > See all xxx apps > Clock > Open)) just for the sake of it but I don't think it's that necessary just yet.

I also did some gaming through Android apps and I'm planning to try Steam for Chromebooks (Beta) too and install some games from my Steam library (primarily Geometry Dash and NFSMW12) but let me tell you something: If you're looking for both gaming performance AND battery life, ChromeOS+Brunch is your way to go. I could get a constant 120 FPS rendering on Asphalt 9 with graphics set to HQ (in case you wonder, my display is 144Hz and ChromeOS could utilize it at its full capability) and my battery lasted for a total of three hours before the OS sent me a low battery notification telling me my battery is estimated to last 15 more minutes before shutdown. I also tried some gaming with osu!lazer inside Crostini and the results were about the same, I've also uploaded a small (HORRIBLE!) playthrough during some debugging process of the audio with someone else who chimed in on the issue I've created at ppy/osu: https://github.com/ppy/osu/discussions/28366#discussioncomment-9624201 (This takes you directly to the comment of mine where I've found a definite solution to it with the help of sineplusx and also uploaded the gameplay with frame graphs enabled.)

Here's a tip to get the most perfect balance of battery life and performance depending on the time of the day: Make sure to set the brightness consistently depending on the brightness of your environment (having too high brightness at night could make your eyes suffer and having too low brightness under sunlight or in a well lit environment can give you some difficulty seeing your screen at all) and set dark mode and night light to turn on and off automatically on sunrise and sunset - "Auto" theme option in Wallpaper & Style and Night light program set to "From sunset to sunrise" in display settings.

RGB keyboard users like me, turning the RGB lighting off completely has ABSOLUTELY NO EFFECT on battery life from my personal observations and doing so even then has the potential to decrease productivity from time to time in a poorly lit environment such as a dark room.

That's all I'm going to say for now, I'll also see ways to squeeze more battery life out of this thing without trimming out some functionalities I use all the time.