sebanc/brunch #361: Can no longer login after switching to dev and updating to 85

KonstantinosDetsis, 2020-06-29:
As mentioned by sebanc [here](https://github.com/sebanc/brunch/issues/345#issuecomment-650987689) to be able to use chrome://flags you need to update to dev channel. I enabled options=enable_updates and rebooted. I then opened up crosh and typed "sudo update_engine_client --channel=dev-channel -update" and waited for it to finish. I rebooted and waited for rootfs to rebuild and once it booted I typed my password and hit enter. It got stuck there. Rebooting and trying again does the same thing. Guest mode doesn't work either. I tried downgrading back to 83 from the developer console but the same thing happened. Any way to fix this without powerwashing or losing my files?

KonstantinosDetsis, 2020-06-29:
I have reinstalled ChromeOS so I will close this issue.

sebanc, 2020-06-29:
It would actually be difficult, if your files are very important you could try to:
- download the samus r83 recovery image from another computer and put it on a usb 
- at login screen type CTRL+ALT+F2 and login as "root"
- mount your usb device and run "chromeos-update -r <path_to_the_samus_r83_recovery_image>"

However, there is no guarantee it would work.

KonstantinosDetsis, 2020-06-29:
As I said I have already downgraded to r83 and I still cannot login. It simply hungs up when pressing enter and pressing the guest mode button doesn't do anything

sebanc, 2020-06-29:
Ah sorry missed that, unfortunately the data partition is encrypted and it's not possible to recover the data as far as I know.

KonstantinosDetsis, 2020-06-29:
Well I don't have important data on my user but I have installed crouton and Ubuntu 18.04 which took me a while to get working. I was originally going to copy the chroot but I found that is was 15GB. Also I already knew that ChromeOS encrypts the user folder after some research. Thanks for your help again.

KonstantinosDetsis, 2020-06-29:
Oh and also why isn't dev 85 working?

closed 2020-06-29 by KonstantinosDetsis

sebanc, 2020-06-29:
That's also what I would like to know, I have already seen the login issue but only when the recovery image was changed through a manual update.

I would be interested to know if you can reproduce the issue on a fresh install.

KonstantinosDetsis, 2020-06-29:
Well before I found [this](https://github.com/sebanc/brunch/issues/165#issuecomment-630917850) I was originally using rammus but then used chromeos-update to switch to samus. However when the update was done it kept telling me that my password was wrong so I had to enter my google password and reset the user.

KonstantinosDetsis, 2020-06-29:
Here is a video of what is happening. (Sorry for low quality) https://drive.google.com/file/d/1EAozHGmUI0_hpVU2-9cTnYdYnph14Q9D/view?usp=sharing

KonstantinosDetsis, 2020-06-29:
Should I reinstall ChromeOS now?

sebanc, 2020-06-29:
Sorry but I have no idea how to recover from this...

KonstantinosDetsis, 2020-06-29:
Ok, I will reinstall tomorrow and see if it happens again on a clean install.

KonstantinosDetsis, 2020-06-30:
I have reinstalled ChromeOS but now when I try to switch to dev channel this happens:
chronos@localhost / $ sudo update_engine_client --channel=dev-channel -update
[0630/113506.859618:INFO:update_engine_client.cc(428)] Channel permanently set to: dev-channel
[0630/113506.859861:INFO:update_engine_client.cc(478)] Forcing an update by setting app_version to ForcedUpdate.
[0630/113506.859885:INFO:update_engine_client.cc(480)] Initiating update check.
[0630/113506.862532:INFO:update_engine_client.cc(509)] Waiting for update to complete.
[0630/113537.725366:ERROR:update_engine_client.cc(210)] Update failed, current operation is UPDATE_STATUS_IDLE, last error code is ErrorCode::kOmahaErrorInHTTPResponse(37)

Any ideas?

sebanc, 2020-06-30:
It seems updates are not correctly enabled in the kernel command line.

KonstantinosDetsis, 2020-06-30:
Oh yeah I forgot.

KonstantinosDetsis, 2020-06-30:
I successfully updated and chrome://flags works now.

sebanc, 2020-06-30:
Nice to see that it eventually worked :)