sebanc/brunch #362: Potential to direct-boot 'chromeos.img' file using GRUB2?

darethehair, 2020-06-29:
Background: I very much use and enjoy the feature of GRUB2 to directly boot ISO images on USB sticks -- giving simple ability to 'pick' the ISO I want to boot from the 'menuentry' system that it provides i.e. no need to fully install the Linux distro that I wish to run.

It seems to me that Brunch 'chromeos.img' files might have the same potential, seeing as the instructions for 'dual-boot' already provide something very close to this e.g. on my computer:

```
menuentry "ChromeOS (boot from disk image)" {
	img_part=/dev/sda4
	img_path=/chromeos.img
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
		cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
	initrd (loop,gpt7)/initramfs.img
}
```
However, my efforts to use a GRUB2 config file similar to this on my USB stick (containing 'chromeos.img') by tweaking the 'img_part' and 'img_path' variables results in the boot process not being able to 'find' ChromeOS (on 'dev/sda and /dev/sdb').

Here is what one of my typical ISO Linux distro GRUB2 menuentries look like:

```
menuentry "Mint Debian Edition 4 Cinnamon 64-bit" {
    set isofile="/boot/isos/lmde-4-cinnamon-64bit.iso"

    loopback loop $isofile
    linux (loop)/live/vmlinuz findiso=${isofile} debug boot=live
    initrd (loop)/live/initrd.lz
}
```
Any ideas on how I can help Brunch to 'find' and use my 'chromeos.img' file?  BTW, I am assuming that I can treat this as I would an ISO file -- if not, I could use utilities to 'convert' it from IMG to ISO, I suppose.

sebanc, 2020-06-29:
Hi,

In general it should work exactly as you mention by just tweaking the 'img_part' and 'img_path' variables.

A few things which might explain it not booting:
- Your usb disk is booted in mbr mode.
- The partition type where the brunch image is located is not ext4.

Could you post the output of "ls" command in your usb grub ?

darethehair, 2020-06-29:
Not quite sure what you are asking for, but here goes:
```
darren@frodo3:/data/download/isodirs/chromeos$ sudo fdisk -l /dev/sdb
Disk /dev/sdb: 28.9 GiB, 30959075328 bytes, 60466944 sectors
Disk model: STORE N GO      
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: F993CF0C-5B7C-4933-9C5F-B430BA97C451

Device      Start      End  Sectors  Size Type
/dev/sdb1    2048     4095     2048    1M BIOS boot
/dev/sdb2    4096   208895   204800  100M EFI System
/dev/sdb3  208896 60466910 60258015 28.8G Linux filesystem

darren@frodo3:/data/download/isodirs/chromeos$ df
Filesystem     1K-blocks      Used Available Use% Mounted on
udev             6056444         0   6056444   0% /dev
tmpfs            1217176      9520   1207656   1% /run
/dev/sda2       51343840  12425296  36280720  26% /
tmpfs            6085872    151244   5934628   3% /dev/shm
tmpfs               5120         4      5116   1% /run/lock
tmpfs            6085872         0   6085872   0% /sys/fs/cgroup
tmpfs             204800      1032    203768   1% /var/log
tmpfs            6085872         0   6085872   0% /var/tmp
tmpfs             102400     56724     45676  56% /tmp
/dev/sda3      824650152 521076788 261613940  67% /DATA
tmpfs            1217172        72   1217100   1% /run/user/1000
/dev/sdb3       29524832   9992660  18009340  36% /media/darren/LINUX

darren@frodo3:/data/download/isodirs/chromeos$ ls -ltr chromeos.img
-rw-r--r-- 1 root root 15032385536 Jun 29 08:46 chromeos.img

darren@frodo3:/data/download/isodirs/chromeos$ ls -l /media/darren/LINUX/boot/isos/chromeos.img
-rw-r--r-- 1 root root 15032385536 Jun 29 09:50 /media/darren/LINUX/boot/isos/chromeos.img
```

So my 'chromeos.img' file is the same size on the HD and the place I copied it to on my USB stick (where I typically put my Linux ISO files).  Yes, I typically had used 'vfat' as the partition type for my ISOs, but the 4gb limit was an obstacle for 'chromeos.img', so I went for 'ext4'.

sebanc, 2020-06-29:
Sorry I was not clear, when you boot from usb and grub appears, press "c" and type "ls", then take a photo.

Aside from that, from what i can tell, you should use:
img_part=/dev/sdb3
img_path=/boot/isos/chromeos.img

darethehair, 2020-06-30:
Good news!  You suggested variables worked great!  Thanks!  Ultimately, with a large-enough USB stick, one could theoretically have multiple Brunch images to pick from (ramus, sammus, bios, etc. flavors).  In my case, it would be nice to have at least one so that I could potentially boot/test/install Brunch on a given destination computer.

A few other related thoughts/questions...

1) The chromeos-install.sh 'dst' parameter seems to react to locations differently depending on the way the destination is referenced i.e. 'chromeos.img' works fine, but (for example) '/media/darren/ISO/boot/isos/chromeos.img' does not.  Not sure if there is something strange about my setup, but all it meant was that I did this in two steps -- create IMG file, and then copy it myself to my USB stick.  No big deal :)

2) As above, your suggested location variables worked just fine.  It makes me wonder, though, why I can use a non-device specific location reference like "/boot/isos/lmde-4-cinnamon-64bit.iso" for my Linux ISOs, but for Brunch/ChromeOS I need to specifically mention the actual device.  Any way of getting around this?  Again, no big deal but just curious :)

3) Brunch is totally amazing, and it 'just works', but I wonder what/how it works its magic.  For example, in my case, a 15gb 'chromeos.img' file boots from a USB stick, and takes some initial effort 'rebuilding ChromeOS rootfs'.  What is happening inside my 15gb file when this happens?  Later, when I sign into my Google account, time is taken to install (?) my apps/extensions.  What if the 15gb 'chromeos.img' file on my USB stick is not large enough?  Maybe this means that a 'minima' account should be used for demoing/installing Brunch from a USB stick to avoid this risk?

4) Though I have not tried it (yet) in this new specific case, after 'demoing' Brunch on a computer using this USB technique, I then might want to presumably 'install' it for real, using a command like 'sudo chromeos-install -dst /dev/sdx'.  This is very simple for me, but is there any easy potential for Brunch to eventually have a nice CLI script or GUI app for Linux beginners, to make it as easy to install as a Linux distro?

kowalski7cc, 2020-06-30:
Maybe something like Android-X86 installer?
https://www.android-x86.org/installhowto.html

darethehair, 2020-06-30:
> Maybe something like Android-X86 installer?
> https://www.android-x86.org/installhowto.html

Yes, perhaps, except rather that forcing the 'install' choice when first booting, I would want to be able to 'demo' Brunch to make sure everything works nicely, and then click an 'icon' (or whatever) to install via GUI.  If a CLI script instead, it would safely show me the partitions available -- with a suggested one.  Of course being able to adjust partitions would be nice too -- but that is a LOT more complicated.

sebanc, 2020-07-01:
> Good news! You suggested variables worked great! Thanks! Ultimately, with a large-enough USB stick, one could theoretically have multiple Brunch images to pick from (ramus, sammus, bios, etc. [...]

That's weird, I tried brunch installer with much more complicated paths (longer / with spaces), not sure what could be the issue with your path as it seems quite standard.

> 2. As above, your suggested location variables worked just fine.  It makes me wonder, though, why I can use a non-device specific location reference like "/boot/isos/lmde-4-cinnamon-64bit.iso" for my Linux ISOs, but for Brunch/ChromeOS I need to specifically mention the actual physical device.  Any way of getting around this?  Again, no big deal but just curious :)

I believe they use an auto-detection method for the partition, I could probably do that but it would mean mounting all partitions to scan for the file which would increase even more the boot delay.

> 3. Brunch is totally amazing, and it 'just works', but I wonder what/how it works its magic.  For example, in my case, a 15gb 'chromeos.img' file boots from a USB stick, and takes some initial effort 'rebuilding ChromeOS rootfs'.  What is happening inside my 15gb file when this happens?  Later, when I sign into my Google account, time is taken to install (?) my apps/extensions.  What if the 15gb 'chromeos.img' file on my USB stick is not large enough?  Maybe this means that a 'minimal' account should be used for demoing/installing Brunch from a USB stick to avoid this risk?

Rebuilding rootfs just means copying partition 5 (ROOTB which is the untouched chromeos filesystem) over partition 3 (ROOTA) and applying all userspace patches.

Brunch minimum size is 14gb to ensure that a minimum of 4gb is available for data. ChromeOS probably knows how to handle the too much apps for storage situation as apps are synced between devices and some chromebooks only have 16gb storage.

> 4. Though I have not tried it (yet) in this new specific case, after 'demoing' Brunch on a computer using this USB technique, I then might want to presumably 'install' it for real, using a command [...]

Two things here:
- I don't know how to write chromeos apps, I just build the kernel and add some shell scripts.
- I am not sure it would even be possible without building chromeos from source (which is not the way brunch works)

darethehair, 2020-07-01:
Thanks for the answers!

closed 2020-07-01 by darethehair