sebanc/brunch #363: Wi-Fi not working

Shershshsh, 2020-06-29:
Hi. Sorry for bother you again. One week ago i created issue here that my rtl8821ce isnt working and options=rtl8821ce helped me. But today i installed triple boot system (win10 linux mint and chrome) and wi-fi not working again even with options. Besides my removable wifi adapter can see networks but cant connect them. When i boot windows or linux wifi is ok. I attach dmesg file
[dmesg.txt](https://github.com/sebanc/brunch/files/4846947/dmesg.txt)

sebanc, 2020-06-29:
Hi,

The options should be in the same variable and coma separated: "options=enable_updates,rtl8821ce"

Shershshsh, 2020-06-30:
Thanks, it works now. It’s strange that last time everything worked with two options

closed 2020-06-30 by Shershshsh

vinoddigital, 2020-07-01:
Realtek RTL8723DE WLAN Driver not working. Hardware HP 15 DA0073TX , No wifi network available after boot through chrome OS. My system is triple boot only Linux mint and windows is working in wi-fi.