sebanc/brunch #365: Chrome os taking much time load apps

debendrakumar, 2020-06-30:
Issue is that when we boot up chrome os  start  and  logged in and tap any apps from the playstore , it says loading the app and take about 2 min to 3 min to start the app
Some time it doesn't  start the app and keep loading .. This problem is on every boot of chrome os ...   
(Lap info HP 15  notebook , installed on hdd , i3 7th gen processor, 4gb ram )

sebanc, 2020-07-01:
Hi, which brunch version do you use and which recovery ?

debendrakumar, 2020-07-01:
The latest testing version  in which  rtl8723de support was given that one
Okay
But some time it take much more time  and i have to wait so much like 10 to 15 min bout still persist loading   and when i restart  once or twice get loaded in 2 or 3 min ....

sebanc, 2020-07-01:
I post it a lot so you have probably already done it but in case you have not done it you can try deleting google sync ?
https://support.google.com/chromebook/answer/2914794?hl=en

sebanc, 2020-07-02:
I included a patch which seems to help with android apps (from issue #271) in the new unstable build:
https://github.com/sebanc/brunch-testing/releases

Could you try it ?

debendrakumar, 2020-07-02:
Ok i will try it

debendrakumar, 2020-07-02:
Sir can you give support for the bluetooth in the rtl8723 d devices

debendrakumar, 2020-07-02:
Please if can ..

debendrakumar, 2020-07-02:
When updated the brunch releases every thing stoped man on login page not able to login now

debendrakumar, 2020-07-02:
Anyidea to go backon on last brunch release  without  uninstalling chromeos...?

debendrakumar, 2020-07-02:
Please i dont want to losse my Apps and data.

sebanc, 2020-07-02:
If you still have the old version, you can do CTRL+ALT+F2 at log screen and update back to it.

debendrakumar, 2020-07-02:
[image]
[image]
[image]
[image]
[image]

These are some pics after updating the testing unstable version

debendrakumar, 2020-07-02:
How to do in developer console i dont ?

debendrakumar, 2020-07-02:
i dont any thing about the developer console

sebanc, 2020-07-02:
try pressing "e" when you see grub and add "options=rebuild" or add "rebuild" to the options if you already have some.

debendrakumar, 2020-07-02:
Ok this will lead to which path ?

sebanc, 2020-07-02:
Potentially it will fix your issue. Else it means that the chromeos-update failed for some reason I don't understand.

sebanc, 2020-07-02:
If you manage to get to login screen and login still does not work, press CTRL+ALT+F2, login as "root" and run chromeos-update with a previous brunch archive.

debendrakumar, 2020-07-02:
Ok thanks but after adding rebuild in thesame error occured "tar:something
something

sebanc, 2020-07-02:
ok it means that the chromeos-update did not work as I tested it on my computer I don't have the tar... isuue (it is not hardware dependent).

If you manage to get to login screen and login still does not work, press CTRL+ALT+F2, login as "root" and run chromeos-update with a previous brunch archive.

debendrakumar, 2020-07-02:
What to do after this one

debendrakumar, 2020-07-02:
[image]

debendrakumar, 2020-07-02:
Actually my archieve is in dowload folder in chomebook

sebanc, 2020-07-02:
I have no idea what you are trying to do here.

debendrakumar, 2020-07-02:
Actually how to update the brunch ..
And my previous brunch release  is in chromebook can i access it or not..

sebanc, 2020-07-02:
No you cannot acces userdata before login. use chromeos-update with the latest brunch release (not unstable)

debendrakumar, 2020-07-02:
Then how to update the brunch now

debendrakumar, 2020-07-02:
By external usb card or anything

debendrakumar, 2020-07-02:
Can give a quick tutorial for this please

sebanc, 2020-07-02:
- Download brunch standard release on a usb flashdrive / sdcard from your phone or anything.
- Boot your laptop and go to the console (CTRL+ALT+F2), login as "root"
- mount your usb flashdrive / sdcard: `mount /dev/sdX /mnt`
- Use `chromeos-update -f /mnt/<path_to_brunch>`

Between, even though I don't think the new release has an issue (something must have gone wrong with chromeos-update), you should not install something called brunch-testing / unstable on a laptop which contains sensitive data.

Try it on usb next time.

debendrakumar, 2020-07-02:
Thanks sir

debendrakumar, 2020-07-02:
mount /dev/sdb/mnt
mount: /dev/sdb/mnt: can' t find in /etc/fstab

This came during mounting san disk  cruser blade

debendrakumar, 2020-07-02:
[image]

sebanc, 2020-07-02:
"mount /dev/sdb /mnt"

debendrakumar, 2020-07-02:
[image]

debendrakumar, 2020-07-02:
[image]
[image]

debendrakumar, 2020-07-02:
[image]
[image]

sebanc, 2020-07-02:
"mount /dev/sdb1 /mnt"

debendrakumar, 2020-07-02:
All possibilitie i have tried no one worked

debendrakumar, 2020-07-02:
Worked

debendrakumar, 2020-07-02:
Now worked sdb1

debendrakumar, 2020-07-02:
Thanks sir for that much effort ....
So much thank you

sebanc, 2020-07-04:
Android apps should work better in the new release I have just uploaded, You should probably backup your data and perform a clean install.

debendrakumar, 2020-07-04:
We have to clean install....

debendrakumar, 2020-07-07:
actually this version also has this problem 
And when  i enable the option of internal drive it get stucked on loading apps for long time

sebanc, 2020-07-07:
Could you try to add "options=original_init" to the kernel command line ?