sebanc/brunch #367: How to enable silent boot?

NovaTheMighty, 2020-07-01:
I want to enable a completely silent boot without all the messages that occur after GRUB boots into Chrome OS. How do I do that?

b1t0y, 2020-07-01:
> I want to enable a completely silent boot without all the messages that occur after GRUB boots into Chrome OS. How do I do that?

Put "console="  after "cros debug" and before the "options=" in kernel command line.

PanKarLos, 2020-07-02:
Can you be more specific ? I'm kinda newbie in this, don't even know where to locate kernel command line. Thank you

b1t0y, 2020-07-03:
So on grub menu when you boot press E to edit.   You will see cros_debug from there.  There is where you will type console= then press ctrl-x of F10 to boot.   If you like to edit the grub from chrome OS itself.  Follow the one on the github readme.  Hope this helps

CarlitoisLoco, 2020-07-06:
so, [..] cros debug console= options=enable_updates [...]?
or is it "console=1"?

Crescendo-BLYAT, 2020-07-06:
[..] cros debug console= options=enable_updates [...]

is the correct one...