sebanc/brunch #369: Issues with Webcam - Intel RealSense 3D Camera F200

kkondath, 2020-07-03:
Hi, I tried installing latest rammus and nami build for version 87 and in both cases the webcam was only showing the depth mode of the 3D camera and not showing up RGB version of the same. Looks it is only installing /  detecting one camera. It should detect 2 of them. It only list 1 camera everywhere (the depth one) which only shows a Greenish image.  

I tried installing the Chromium OS build directly (not Chrome OS) and when I use that, 2 Webcams show up in all apps including Camera (Default being RGB and I can switch to Depth if needed) and so it works without a problem....

The challenge with Chromium OS build is that it does not support Family Link accounts and that feature works fine with Chrome OS builds. I am now left with not being able to use either OS since one of my main needs is for using distance learning (webcam) with a managed account (Family  Link)

Laptop - Dell Inspiron 15 i5559-4682SLV Signature Edition Laptop
• 15.6-inch Full HD touchscreen
• Intel Core i5-6200U
• 8GB memory
• Intel RealSense 3D camera 

Full Camera Name:
Intel(R) RealSense(TM) 3D Camera (Front F200)

sebanc, 2020-07-03:
Hi, rammus and nami both use ChromeOS new camera framework which happens to be less compatible than the old one. I am trying to improve it but without success for now.

Nevertheless, "reef" image has the old camera framework and should work with your cameras, could you try it ?

kkondath, 2020-07-03:
Sure sebanc, trying with reef now...will post back in a few hours

kkondath, 2020-07-03:
Thanks sebanc, works fine with "reef"... are there any major differences in "reef" vs "rammus", everything seems to work fine for me, just checking

sebanc, 2020-07-04:
No, they are the same aside the camera framework as far as I know. Historically rammus was the recommended image because it was compatible with more devices but since r83 it is the same as reef. Nevertheless as rammus will have longer updates support, I still maintain rammus as the recommended image hoping that at some point the camera issue can be fixed.

kkondath, 2020-07-04:
Ran into an issue with "reef",  microphone does not work...

I have Realtek Audio Microphone (built in), any suggestions

sebanc, 2020-07-05:
Did your microphone work with other recovery images ?

The microphone is a quite common issue which is generally independent of the recovery image. There are some workarounds available in the issue #56.

kkondath, 2020-07-05:
Yup, not working in rammus as well, just tried it..... switching back to reef and trying workarounds.

kkondath, 2020-07-05:
Workaround worked.... will continue using for a few days on USB boot and then move to HDD dual boot

closed 2020-07-05 by kkondath