sebanc/brunch #371: Power button does not work

jhedfors, 2020-07-04:
I am running this on a Dell inspiron 15 3567 laptop. Is tapping the power button supposed to do something? 

I would expect that a tap would put it to sleep/wake it up and a longer press would show a power menu.

sebanc, 2020-07-04:
Hi,

Indeed this is the standard behavior but your device probably has acpi power buttons, try adding "options=acpi_power_button" to the kernel command line. There is no long press for those, it will just display the power menu when the power button is pressed.

jhedfors, 2020-07-04:
That did the trick! Thanks for the help!

closed 2020-07-04 by jhedfors

bybenjamin, 2022-12-22:
I'm having the same problem. Adding acpi_power_button doesn't work.
[dmesg.txt](https://github.com/sebanc/brunch/files/10288357/dmesg.txt)
[dmesg_power.txt](https://github.com/sebanc/brunch/files/10288359/dmesg_power.txt)
[evtest.txt](https://github.com/sebanc/brunch/files/10288360/evtest.txt)